roleoroleo/sonoff-hack #1: ONVIF Network Interface not persistent

barrymossel, 2020-09-12:
When changing the ONVIF Network Interface via the web interface to Wireless, press save and refresh the page, it goes back to ethernet.

roleoroleo, 2020-09-12:
It's a typo.
You can fix it changing ONVIF_INTERFACE with ONVIF_NETIF inside 2 files:

- /mnt/mmc/sonoff-hack/etc/system.conf
- /mnt/mmc/sonoff-hack/script/check_conf.sh

barrymossel, 2020-09-12:
Cool will try it later. Unfortunately ONVIF seems (through ethernet) very slow though...

roleoroleo, 2020-09-12:
What is slow?
ptz?

barrymossel, 2020-09-12:
No, actually the stream is lagging 2-3 seconds (or even more). Big difference opposed to the Ewelink app (or RTSP stream from original firmware - at least it seems). Will test later.

roleoroleo, 2020-09-12:
Onvif it's only a descriptor, doesn't provide stream.
The stream is the original rtsp produced from the cam through the sonoff firmware.

barrymossel, 2020-09-12:
I already had that feeling lol. Just seemed that way then. App is way faster though.

But thanks for this hack. Really nice!

roleoroleo, 2020-09-17:
Fixed in 0.0.5
https://github.com/roleoroleo/sonoff-hack/releases/tag/0.0.5

closed 2020-09-19 by roleoroleo