roleoroleo/sonoff-hack #3: ONVIF error

codebullfrog, 2020-09-17:
Hi,

When I try to add the camera using ONVIF I get an error. In one app I'm using it says "Error in retrieving media profiles.  This could be due to an erroneous media service port." In Home Assistant I get Unknown error occurred and in the logs it says invalid URL with "http://:80".
I can provide more detailed logs if required.

roleoroleo, 2020-09-17:
Did you select the onvif interface in the configuration tab?
Check if it's the same problem discussed here:
https://github.com/roleoroleo/sonoff-hack/issues/1

codebullfrog, 2020-09-17:
I ssh'd in and changed the two files to read ra0 instead of eth0 but stiff have the same issue.

roleoroleo, 2020-09-17:
Which ONVIF client are you using?
Client and cam, are they in the same subnet?
Is the cam connected through the wifi?

codebullfrog, 2020-09-17:
Trying with Blue Iris on the PC and Onvifer on the phone. I have also tried with Home Assistant adding as an integration. They are all on the same subnet and I can see the RTSP streams on all devices.
The cam is connected though WiFi, all my devices seem to do some communication with the cam before having an error.

I have just booted up my other cam for the first time and I am getting the same issues.

roleoroleo, 2020-09-17:
If the url is http://:80 the cam cannot transform the interface name into the IP address.
Could you post your config?

codebullfrog, 2020-09-17:
[config.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/5239529/config.tar.gz)
Thats my config download from the web UI.

roleoroleo, 2020-09-17:
Open a shell and edit the file system.conf.
Change ONVIF_INTERFACE with ONVIF_NETIF.

codebullfrog, 2020-09-17:
I was just looking at that code in system.sh.
I made the change and looks like it is working now.

roleoroleo, 2020-09-17:
Fixed in 0.0.5.
https://github.com/roleoroleo/sonoff-hack/releases/tag/0.0.5

closed 2020-09-19 by roleoroleo