roleoroleo/sonoff-hack #4: Feature request: Disable built in voice - "please wait for internet connecting"

cjonhammar, 2020-09-18:
Using it as a babycamera, and when we have power outage etc the baby gets awakened by that female voice who is letting the baby know that the camera is connecting to internet. :)

roleoroleo, 2020-09-18:
Not so easy.
The audio files are played before the hack.
And the file system is a read only squashfs.
I can't overwrite the files without flashing the whole partition.

alekslyse, 2020-10-20:
You can actually change it /mnt/mtd/ipc/cfg/En.wav - change the wav file to something with no sound. If I remove that file it talk in chinese, even during reboot so the changes are if you put a silent file there you have no voice

roleoroleo, 2020-10-20:
And what about the 2nd wav when the connection is done?

EpicLPer, 2020-10-20:
Thing is, the WAV file @alekslyse mentioned isn't really a WAV file. It's simply a renamed text file.

The real WAV files are in `/mnt/mtd/ipc/app/res/En` but they can't be replaced cause the file system is read-only.  
Maybe you can make an option script to disable those or replace them with custom sounds @roleoroleo? I wasn't able to figure out how to replace files or mount files from the SD to replace the ones in the read-only memory on bootup.

roleoroleo, 2020-10-20:
We could try to turn the volume down.

alekslyse, 2020-10-20:
yes sorry my fault for not checking. I just did see a wav file and assumed it was the voice. sorry for that

roleoroleo, 2020-10-23:
Done:
https://github.com/roleoroleo/sonoff-hack/commit/b9e2852789acebe33e4639bf74c24235e61e335a

EpicLPer, 2020-10-23:
Gave it a try and it indeed works, nice!  
Trying to figure out how to replace the sounds now since the SD Card seems to only get mounted pretty late into the whole boot process :(

EpicLPer, 2020-10-23:
Managed to do it now! Had to mount the SD Card directly in the boot.sh for a short while, then copy the files over, then umount it again. Works now :D 

https://twitter.com/EpicLPer/status/1319674142036283392

roleoroleo, 2020-10-23:
The sd is mounted in a temporary folder (/var/sdcard if I remember correctly).
You could copy the file from /var/sdcard to /tmp and then use mount --bind.

roleoroleo, 2020-10-31:
https://github.com/roleoroleo/sonoff-hack/commit/b9e2852789acebe33e4639bf74c24235e61e335a

cjonhammar, 2021-01-08:
Working fine in 0.09 Thanks a bunch!

closed 2021-01-08 by cjonhammar

rolphan-del, 2026-04-15:
Hello guys, I would be very interested in this option, I have 5 of these cameras and we are scared to death at night when there is a power outage. Where can I find a detailed description that even a very beginner could do?