roleoroleo/sonoff-hack #5: Bug in 0.0.6: All preset combo boxes are blank

KhurramFHassan, 2020-09-19:
I have recently upgraded to version 0.0.6 and all the preset combo boxes in the PTZ page are blank.

roleoroleo, 2020-09-19:
Wich combo?
Go, boot or set?

KhurramFHassan, 2020-09-19:
All combos.

roleoroleo, 2020-09-19:
Please clear the cache of your browser and try again.

KhurramFHassan, 2020-09-19:
I did that on google chrome and it has made no difference. I tried it on another laptop (in chrome and firefox) but still the combos are blank.

roleoroleo, 2020-09-19:
Please check the console of the browser for errors.
Try to execute manually the presets web service:
http://192.168.1.213:8080/cgi-bin/get_configs.sh?conf=ptz_presets
and post the output.

KhurramFHassan, 2020-09-19:
The above link gives

> SyntaxError: JSON.parse: bad control character in string literal at line 2 column 20 of the JSON data

The raw json gives

> {
"0":"empty|1649|147
",
"1":"empty|-1|-1
",
"2":"empty|-1|-1
",
"3":"empty|-1|-1
",
"4":"empty|-1|-1
",
"5":"empty|-1|-1
",
"6":"empty|-1|-1
",
"7":"empty|-1|-1
",
"8":"empty|-1|-1
",
"9":"empty|-1|-1
",
"NULL":"NULL"
}

I dont know where the NULL line is coming from. It is not in the ptz_presets.conf in the backup configuration file.

roleoroleo, 2020-09-19:
Did you edit the file from a Windows machine?
It seems there are wrong CR LF.

KhurramFHassan, 2020-09-19:
Yes I edited it on a windows pc (in 7-zip builtin editor). I will reset to defaults and start again. Also, it seems that the camera reboots when I am playing around with the presets in the PTZ page. Could it be related to windows vs linux CR LF?
Thanks

roleoroleo, 2020-09-19:
Yes it's possible.
But it could also be a bug.

KhurramFHassan, 2020-09-19:
I reset to defaults and started again. I am trying to create a starting preset at slot 0; I called it "startup". To name and save it, I did the following:
1) Move the camera to the desired position using the arrow keys on the page.
2) Fill a name in the "Preset Name" textbox.
3) Choose a preset slot in "Preset Number" combo at the bottom of the page.
4) Click on "Set Preset" button.
5) Set this as the starting preset in the "Go to Preset on Boot" part of the page.

Now going to the json link gives ""startup|1649|-61" in the first line. I rebooted the camera but the camera stayed in the default position. I tried to manually move to preset 0, but the camera stayed in the default position. I am stumped here.

Btw, can you add an uptime somewhere in the page. This would be really helpful in knowing if the camera is rebooting randomly.

Thanks

roleoroleo, 2020-09-19:
Did you save configuration after step 5?

A strange thing: you have a negative y coordinate.
Normally the y coordinate is 0<=y<=900

Uptime is in the status page, click the sonoff hack image on the upper left corner.

KhurramFHassan, 2020-09-22:
Sorry for the late reply; I had some family related engagements. I did save the configuration after choosing the starting preset. I don't know about the y coordinate value as I just used the arrow keys to set the camera and then save the preset. My starting preset is 5 "arrow down" and 1 "arrow right" from the default bootup position. So if I have to set the x/y values myself in the configuration backup archive, what values should I use?

roleoroleo, 2020-09-23:
No, you don't have to set manually the coordinates.
You don't have to edit manually the conf file.

Simply it's strange that the cam detects a negative coordinate.
When the cam starts, it goes automatically to the lower right corner and set the 0,0 coordinates.
Then come back to the last position used.
So there should be no negative coordinates.

KhurramFHassan, 2020-09-23:
After the check yesterday, I went to the PTZ page again and tried to save the configuration again in the "Goto preset on boot" part. I got the message that it is saving and then it got stuck on it. I checked back after a few minutes and saw that the camera had rebooted (uptime was 4 mins) and had gone back to the default position even though the above option is set for my custom preset; the json values are "0":"startup|1649|-61",....".

roleoroleo, 2020-09-23:
I don't understand.
I tried now your procedure and I have no problem.
If it's not a trouble, format the sd card and start from scratch.

KhurramFHassan, 2020-09-23:
I believe that I formatted the sdcard in Ubuntu+gparted and then copied your firmware to the card under Windows. I had earlier formatted the card in the ewelink app and it had created an exfat partition which I could not delete in windows. So I had to use Ubuntu+gparted. I will do this again and this time copy it from Ubuntu also.

roleoroleo, 2020-09-23:
Try to format the sd card with a fat32 filesystem.
I usually use 7zip on Windows to decompress the archive in the sd and it works.
The editor probably has problems with CRLFs but the copy works.

KhurramFHassan, 2020-09-23:
Another problem. I have a 64 GB sdcard and disk management does not allow me to create a fat32 partition on the sd card; the only options are exfat and ntfs. Apparently, fat32 partitions larger than 32 GB are not supported in windows. If I can't use the entire 64 GB with fat32, I might as well get a 32 GB card.

KhurramFHassan, 2020-09-23:
I created a 32 GB fat32 partition on the sdcard on windows and extracted the firmware to it using 7zip. But saving the starting preset rebooted the camera. This camera will just not play nicely with the preset commands on the PTZ page.

roleoroleo, 2020-09-24:
Please try to run the command with ssh:
`/mnt/mmc/sonoff-hack/bin/ptz -d -f /mnt/mmc/sonoff-hack/etc/ptz_presets.conf -a set_preset -n 1 -e Test`

KhurramFHassan, 2020-09-24:
Running this command gives strange results for the current position. When the camera boots, the command output is:
```
debug on
Running` action 114
key=0, value=startup|1650|-60
key=1, value=empty|-1|-1
key=2, value=empty|-1|-1
key=3, value=empty|-1|-1
key=4, value=empty|-1|-1
key=5, value=empty|-1|-1
key=6, value=empty|-1|-1
key=7, value=empty|-1|-1
key=8, value=empty|-1|-1
key=9, value=empty|-1|-1
Current position: (1751, 450)
1
```
After I press 4 "arrow downs" and 1 "right arrow", I get the following:
```
debug on
Running action 114
key=0, value=startup|1650|-60
key=1, value=Test|1751|450
key=2, value=empty|-1|-1
key=3, value=empty|-1|-1
key=4, value=empty|-1|-1
key=5, value=empty|-1|-1
key=6, value=empty|-1|-1
key=7, value=empty|-1|-1
key=8, value=empty|-1|-1
key=9, value=empty|-1|-1
Current position: (1650, 44)
1
```
The strange part is if I now do 1 "arrow down", I start getting negative y coordinates:
```
debug on
Running action 114
key=0, value=startup|1650|-60
key=1, value=Test|1751|44
key=2, value=empty|-1|-1
key=3, value=empty|-1|-1
key=4, value=empty|-1|-1
key=5, value=empty|-1|-1
key=6, value=empty|-1|-1
key=7, value=empty|-1|-1
key=8, value=empty|-1|-1
key=9, value=empty|-1|-1
Current position: (1650, -57)
1
```
One more "arrow down" gives:
```
debug on
Running action 114
key=0, value=startup|1650|-60
key=1, value=Test|1751|-57
key=2, value=empty|-1|-1
key=3, value=empty|-1|-1
key=4, value=empty|-1|-1
key=5, value=empty|-1|-1
key=6, value=empty|-1|-1
key=7, value=empty|-1|-1
key=8, value=empty|-1|-1
key=9, value=empty|-1|-1
Current position: (1650, -157)
1
```
And so on. Mind you, the camera is not moving down any further but the y coordinate keeps going in the negative direction.

roleoroleo, 2020-09-24:
Ok, I understand.
I will check it.

KhurramFHassan, 2020-09-25:
I manually edited ptz_presets.conf in the configuration backup to change the xy coordinates to 1650,49 (for preset 0) and then restored the configuration. Now the camera behaved as expected on bootup and the combo boxes in the PTZ page are displaying values ok. The camera uptime is now about 11 hrs which is the longest that I remember.

roleoroleo, 2020-09-26:
I confirm that, when the camera reaches the lower boundary, the variable that contains the coordinates continues to decrease (even if the camera is stopped).
Same thing for the other sides.
And I can't change this beahvior.
When the coordinate is negative probably there is a kernel panic in the ptz driver.
I will add some check.

closed 2020-10-31 by roleoroleo