roleoroleo/sonoff-hack #6: Authentication not accepted for streaming

dehness, 2020-09-19:
Hi,

I can access and change the webui password, however, I can't seem to get the camera to save the authentication credentials for the stream properly.

When I change the password it works for the webui but whenever I try to access the stream it just comes back with invalid user/password, no matter if I try on Mac or iOS.

Home Assistant will add the camera under the onvif integration and pull the snapshots correctly, but the live stream fails to start.

Initially I couldn't access SSH either but I formatted the sdcard and modified the system.conf before inserting it into the camera and that worked for SSH but still not for streaming.

roleoroleo, 2020-09-19:
Which password are you changing?
SSH_PASSWORD is the /etc/passwd password.
PASSWORD is for HTTP, ONVIF and RTSP services.

dehness, 2020-09-19:
I changed both to admin and username to admin so the passwords are:
admin:admin
root:admin

I've tried multiple passwords and also tried without a password, which works fine for the webui but not for the stream.

roleoroleo, 2020-09-20:
2 problems:
- Users admin and rtsp are internal users and you can't use them.
- The links in the status page are wrong.

Try another username.

dehness, 2020-09-30:
Face palm. When I read this reply I was already deep down a rabbit hole attempting to compile my own firmware so you just saved me hours of my life. Thanks mate, that fixed it.

closed 2020-09-30 by dehness