roleoroleo/sonoff-hack #12: Feature request: Send PTZ commands via MQTT

fcsegalla, 2020-10-04:
This would be awesome

roleoroleo, 2020-10-04:
Did you try http method?

fcsegalla, 2020-10-04:
Yes, it worked fine, the problem with http method is that we have to deal with port forwarding in the client's router, with if can send PTZ commands and ask for a snapshot via MQTT we don't need to change the firewall at all

rt400, 2020-10-05:
what the http method ? can you give example?

roleoroleo, 2020-10-05:
@rt400 
Check the updated wiki.
There was an error.

rt400, 2020-10-05:
@roleoroleo not working
i used with this url : http://192.168.1.101:8080/cgi-bin/preset.sh?action=go_preset&num=0&name=door
and i get This site can’t be reached

roleoroleo, 2020-10-05:
I verified in the archive and the file is present.
Verified also in my cam and it's working.

rt400, 2020-10-05:
@roleoroleo sorry..worng IP

closed 2020-10-31 by roleoroleo