roleoroleo/sonoff-hack #14: Poor image quality

rt400, 2020-10-07:
Hi, First of all thanks for your work, wonderful work and I really enjoy it.
Now, recently I see that the quality that the camera survives is really poor and as if colorless. 
For example : 
[image]

Any idea why ??

roleoroleo, 2020-10-31:
Sorry for the delay.
It seems uv components are inverted.
But I don't know how to solve because the snapshot is created from sonoff software.

closed 2021-05-14 by roleoroleo

dehness, 2021-10-01:
Must be a design fault. One of my cameras does the exact same thing both when streaming via RTSP and when viewing in the Itead app. At first I thought it might be a bug with @roleoroleo's firmware but after removing the SDCard and doing a reset the problem remained so it's unaffected by this hack. Interestingly, the colours appear correctly when panning the camera (or moving it physically), but about a second after the movement stops, the colour problem comes back.

If you happen to find a fix for it, please let me know. 👍

rt400, 2021-10-03:
@dehness same problem. also in ewelink app i got this color problem. i dont have and fix for that , its looks like a hardware issue.