roleoroleo/sonoff-hack #15: Feature Request: Rotate

cipherNexus, 2020-10-07:
When cloud is disabled the eWelink configuration is also disabled, so I'd like to be able to rotate the camera from hack configuration interface, what do you think ?

roleoroleo, 2020-10-07:
I would like to add some configuration opions and "rotate" is in the todo list.
But it is not a development that I will do in a short time.

daveharnett, 2021-10-06:
The rotate toggle in the 0.1.1 version doesn't seem to work. Changes to it get persisted but not enforced. Using the rotate toggle in the ewelink app does work (and the toggle on the sonoff-hack webserver reflects the change).

stbuerger, 2022-02-12:
> The rotate toggle in the 0.1.1 version doesn't seem to work. Changes to it get persisted but not enforced. Using the rotate toggle in the ewelink app does work (and the toggle on the sonoff-hack webserver reflects the change).

+1
Can see changes reflected as in 
   `sqlite3 /mnt/mtd/db/ipcsys.db "select c_param_value from t_sys_param where c_param_name=\"mirror\";"`
but this has no effect on rtsp-stream.
Any ideas which program/process should pick up the settings and actually doing the mirroring.

roleoroleo, 2022-02-13:
> Changes to it get persisted but not enforced

Yes, you need to restart the cam if you change the setting through the web interface.
Please, tread this commit:
https://github.com/roleoroleo/sonoff-hack/commit/18bdd992451e0be75d48dfe5b7cead4c6d77d9e9

stbuerger, 2022-02-13:
> > Changes to it get persisted but not enforced
> 
> Yes, you need to restart the cam if you change the setting through the web interface. Please, read this commit: [18bdd99](https://github.com/roleoroleo/sonoff-hack/commit/18bdd992451e0be75d48dfe5b7cead4c6d77d9e9)

PERFECT !
THX a lot for the quick response and fix. I can confirm it works as intended.

closed 2024-05-08 by github-actions[bot]