roleoroleo/sonoff-hack #16: Onvif on synology surveillance station

alekslyse, 2020-10-14:
Have anyone made this work with synology surveillance station onvif. Onvif works on my phone apps so it’s working, but synology say can’t read capabilities. Any suggestions?

Rtsp works in synology, but would need ptz

roleoroleo, 2020-10-14:
It's a known issue.
Please, send me a dump of the network traffic during the negotiation, if you are able.
This is tcpdump:
[tcpdump.gz](https://github.com/roleoroleo/sonoff-hack/files/5376009/tcpdump.gz)

EpicLPer, 2020-10-14:
As a kind of related but somewhat offtopic still reply: Blue Iris was not automatically able to detect the High Quality Stream via the “Find/Inspect” button. Normally afaik that URL should be sent via the ONVIF  discovery, right?

alekslyse, 2020-10-14:
> It's a known issue.
> Please, send me a dump of the network traffic during the negotiation, if you are able.
> This is tcpdump:
> [tcpdump.gz](https://github.com/roleoroleo/sonoff-hack/files/5376009/tcpdump.gz)

Just a tcp dump from the synology to the camera or a general tcp dump whie trying to add the camera

roleoroleo, 2020-10-14:
The traffic from the synology and the camera when you try to add the onvif cam.

alekslyse, 2020-10-14:
Here is a tcpdump 192.168.1.204 is the camera. pasword and username is set as test/test

[test.pcapng.zip](https://github.com/roleoroleo/sonoff-hack/files/5378394/test.pcapng.zip)

alekslyse, 2020-10-14:
For reference I also took a tcpdump on a onvif camera that works. 192.168.1.52 (NOT this hack, just for a reference of something thats working)

[lastestov.pcncap.zip](https://github.com/roleoroleo/sonoff-hack/files/5378687/lastestov.pcncap.zip)

roleoroleo, 2020-10-14:
Thank you.
The 1st capture is ok.
The 2nd one containts only request packets, reponses are missing.

alekslyse, 2020-10-14:
The test.pcapng.zip is the one that doesnt work (error), the second lastestov.pcncap.zip works. The test.pcapng.zip is the sonoff onvif hack

roleoroleo, 2020-10-14:
> As a kind of related but somewhat offtopic still reply: Blue Iris was not automatically able to detect the High Quality Stream via the “Find/Inspect” button. Normally afaik that URL should be sent via the ONVIF discovery, right? But PTZ there works just fine :)

Onvif discovery should send only the generic onvif url, not the streams.

roleoroleo, 2020-10-14:
@alekslyse 
Please try this binary:
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5380701/onvif_srvd.gz)

alekslyse, 2020-10-15:
Progress; It now detects the capabilities, but does not get the stream url. Still .204 ip on the camera. 

[onviftest.pcncap.zip](https://github.com/roleoroleo/sonoff-hack/files/5385203/onviftest.pcncap.zip)
[image]

alekslyse, 2020-10-15:
For reference I did two captures, attached. onviftest2.pcncap with 192.168.1.204 with clicking the button test connection producing no picture

the other file onviftestothercam.pcncap with 192.168.1.52 is a known working onvif camera with clicking the test connection producing picture.

Maybe that will clarify what response the two cameras get 

[Archive 2.zip](https://github.com/roleoroleo/sonoff-hack/files/5385282/Archive.2.zip)

roleoroleo, 2020-10-15:
In the capture file of the working cam I can't find the response packets.
Could you check the capture procedure?

alekslyse, 2020-10-15:
> In the capture file of the working cam I can't find the response packets.
> Could you check the capture procedure?

Just a regular sudo /usr/sbin/tcpdump -w /volume1/tmp.pcncap

Its the same on both files. Any arguments I should add?

roleoroleo, 2020-10-15:
I don't know what the difference.
Try to capture the traffic on the synology, if it's possible.

However I checked the traffic and it seems synology requires dynamic profile creation. This device does not support this function (CreateProfile and DeleteProfile).
And I don't understand why synology should ask for this feature...
In my opinion it is wrong.

alekslyse, 2020-10-15:
its captured on the synology directly. I assume onvif will not work on the synology then?

roleoroleo, 2020-10-15:
Onvif it's not a good standard.
Each manufacturer creates its own implementation.
There are a lot of compatibility problems.

roleoroleo, 2020-10-15:
Another try, if you can...
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5386793/onvif_srvd.gz)

alekslyse, 2020-10-15:
Nothing better. Its a lot of chatter between the cam and the onvif, but when it ask for the add of profile it seems like it gives a 400 bad request back
 
[onvif20.pcncap.zip](https://github.com/roleoroleo/sonoff-hack/files/5387415/onvif20.pcncap.zip)

alekslyse, 2020-10-15:
onvif device manager on windows wont even connect to the camera so something is strange

alekslyse, 2020-10-15:
I see another strange thing; If I reboot the camera, and after the voice; If I use onvif device manager on windows and click refresh I get into the onvif settings when the camera start to turn at bootup. After its done turning no more onvif or able to connect anymore

roleoroleo, 2020-10-15:
The versions I'm sending you are experimental.
There is an alpha implementation of event management.
So it's not stable. Use it only to test the synology.

One more time around the block:
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5387600/onvif_srvd.gz)
I changed the soap fault returned with CreateProfile and DeleteProfile.
I added the GetGuaranteedNumberOfVideoEncoderInstances function.

roleoroleo, 2020-10-15:
If this doesn't work I don't know ho to solve the issue.
Probably I need to create a fake CreateProfile function.

alekslyse, 2020-10-15:
> The versions I'm sending you are experimental.
> There is an alpha implementation of event management.
> So it's not stable. Use it only to test the synology.
>
> One more time around the block: [...]

That seemed to turn off the whole onvif. It could not find the cam on port 80 or load anything

roleoroleo, 2020-10-16:
I will check it.

roleoroleo, 2020-10-16:
I fixed a lot of bugs:
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5394138/onvif_srvd.gz)

alekslyse, 2020-10-17:
No still no connection. another dump here. with the two last releases the onvif communication is never happening.

No much exciting in the log, still with filter ip.src == 192.168.1.204 or ip.dst == 192.168.1.204 - just communication failures (its the correct ip etc - can control through web ui)

Just for a ps I do NOT think synology support onvif 1.0 or 1.1, but those should be very old. I dont know what version this one is using.

[onvif50.pcncap.zip](https://github.com/roleoroleo/sonoff-hack/files/5395745/onvif50.pcncap.zip)

roleoroleo, 2020-10-18:
I tested it again and it works (apart from a memory leak problem).
Try to run it manually with --no_fork and --no_close options.

alekslyse, 2020-10-18:
Im not sure how you mean with running it manually? how?

roleoroleo, 2020-10-18:
Connect to the cam with ssh and run the following command:
```
onvif_srvd --no_fork --no_close --pid_file /var/run/onvif_srvd.pid --model "Sonoff Hack" --manufacturer Sonoff --firmware_ver 0.0.3 --hardware_id GK-200MP2B --serial_num a6300087b8 --ifs eth0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 1920 --height 1080 --url rtsp://hack:hack@%s/av_stream/ch0 --snapurl http://%s:8080/cgi-bin/snapshot.sh --type H264 --name Profile_1 --width 640 --height 360 --url rtsp://hack:hack@%s/av_stream/ch1 --snapurl http://%s:8080/cgi-bin/snapshot.sh --type H264 --ptz --move_left "/mnt/mmc/sonoff-hack/bin/ptz -a left" --move_right "/mnt/mmc/sonoff-hack/bin/ptz -a right" --move_up "/mnt/mmc/sonoff-hack/bin/ptz -a up" --move_down "/mnt/mmc/sonoff-hack/bin/ptz -a down" --move_stop "/mnt/mmc/sonoff-hack/bin/ptz -a stop" --move_preset "/mnt/mmc/sonoff-hack/bin/ptz -f /mnt/mmc/sonoff-hack/etc/ptz_presets.conf -a go_preset -n %t" --set_preset "/mnt/mmc/sonoff-hack/bin/ptz -f /mnt/mmc/sonoff-hack/etc/ptz_presets.conf -a set_preset -e %n -n %t"
```

alekslyse, 2020-10-19:
I just get a segment error

[image]

roleoroleo, 2020-10-19:
Strange behavior.

alekslyse, 2020-10-20:
Yeah. Could it be because I set it up in the app first? The old rtsp stream still works. I tried to reinstall your hack (format sd), but still same issue.

alekslyse, 2020-10-21:
Some progress; Your latest file does not work, it just give segmentation failure. (im using wifi). changed the start file to the wifi and serial just to be sure but still segmentation failure. If your camera works on the latest file your sent; how do you setup your camera? wifi / eth / username / password / setup via app. I even tried to reset the camera with your latest, and still it does not even start the onvif server

I has to set username and password to hack for the profiles to work at all

The second to last where you removed the profile thing it does a bit more now. It accepts the profiles, but errors in "Max number of profiles reached"

I also added the console dump with the ps -l after reset of camera and your latest file as well as wireshark dump where it say profiles reached. If we can bypass the profiles reached I think we are on the right path.

[console.txt](https://github.com/roleoroleo/sonoff-hack/files/5416337/console.txt)
[onvif100.pcncap.zip](https://github.com/roleoroleo/sonoff-hack/files/5416319/onvif100.pcncap.zip)

roleoroleo, 2020-10-21:
The message "Max number of profiles reached." is normal.
It's the response of the cam when the client tries to create a dynamic profile (not supported).
Segfault is not normal...
I will check your capture file.

alekslyse, 2020-10-21:
I have no idea what else to do. I have reset the camera, wifi and ethernet, reset your hack etc. with the last one onvif does not start at all. the previous ones it starts and the command works. Its just the last one

roleoroleo, 2020-10-21:
If you try the cam with odm, does it work?

alekslyse, 2020-10-21:
Not with the last release as it doesn’t start auto or manually. Second to last release it works for 2-3 seconds after the swirling of the camera on boot if I keep clicking connect, then  it doesn’t work anymore in odm. Synology recognises second to last too, but dont get the rtsm stream so cant go beyond the setup screen the no more profile gives a internal error

roleoroleo, 2020-10-21:
I don't understand.
With the last version, does the process crash immediately or after a discovery?
The capture you posted contains a long dialog between the client and the cam and it seems that the cam crashes after the CreateProfile message.
I'm probably confusing the various versions.

alekslyse, 2020-10-21:
Sorry about that. Latest version doesn’t start automatically or by cli (segmentation fault). With that installed its no service running at boot. The latest wireshark is from the second to last. Second to last start automatically and by cli tested both methods

roleoroleo, 2020-10-22:
Try this please:
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5424881/onvif_srvd.gz)

alekslyse, 2020-10-22:
same error. no onvif server started on camera boot, and manual give segment failure. It has to be something that got removed from now the third to last (as the third to last it booted the onvif server etc)

roleoroleo, 2020-10-22:
Try to run it with strace:
[strace.gz](https://github.com/roleoroleo/sonoff-hack/files/5425294/strace.gz)

roleoroleo, 2020-10-23:
Try to update all with new 0.0.7

alekslyse, 2020-10-23:
I start to wonder if its something wrong with the camera. You guys are running stock cameras or have you done any firmware hacks / opened the cameras before doing this? I assumed it was just uploading it to the camera and boot it (even tried to reset it, plus using ethernet)

Even with .7 onvif wont start at boot. Here is the ps list right after boot

`[root@GK]# ps -l
S   UID   PID  PPID   VSZ   RSS TTY   STIME TIME     CMD
S     0     1     0  1448   336 0:0   23:38 00:00:01 {squashfs_init} /gm/bin/busybox ash /squashfs_init
S     0     2     0     0     0 0:0   23:38 00:00:00 [kthreadd]
S     0     3     2     0     0 0:0   23:38 00:00:00 [ksoftirqd/0]
S     0     4     2     0     0 0:0   23:38 00:00:00 [kworker/0:0]
S     0     5     2     0     0 0:0   23:38 00:00:00 [kworker/u:0]
S     0     6     2     0     0 0:0   23:38 00:00:00 [khelper]
S     0     7     2     0     0 0:0   23:38 00:00:00 [sync_supers]
S     0     8     2     0     0 0:0   23:38 00:00:00 [bdi-default]
S     0     9     2     0     0 0:0   23:38 00:00:00 [kblockd]
S     0    10     2     0     0 0:0   23:38 00:00:00 [khubd]
S     0    11     2     0     0 0:0   23:38 00:00:00 [kworker/0:1]
S     0    12     2     0     0 0:0   23:38 00:00:00 [cfg80211]
S     0    13     2     0     0 0:0   23:38 00:00:00 [kworker/u:1]
S     0    14     2     0     0 0:0   23:38 00:00:00 [kswapd0]
S     0    15     2     0     0 0:0   23:38 00:00:00 [ksmd]
S     0    16     2     0     0 0:0   23:38 00:00:00 [fsnotify_mark]
S     0    17     2     0     0 0:0   23:38 00:00:00 [crypto]
S     0    24     2     0     0 0:0   23:38 00:00:00 [mtdblock0]
S     0    25     2     0     0 0:0   23:38 00:00:00 [mtdblock1]
S     0    26     2     0     0 0:0   23:38 00:00:01 [mtdblock2]
S     0    27     2     0     0 0:0   23:38 00:00:00 [mtdblock3]
S     0    28     2     0     0 0:0   23:38 00:00:00 [mtdblock4]
S     0    29     2     0     0 0:0   23:38 00:00:00 [mtdblock5]
S     0    32     2     0     0 0:0   23:38 00:00:01 [mmcqd/0]
S     0    33     2     0     0 0:0   23:38 00:00:00 [deferwq]
S     0    53     1  1448   320 0:0   23:38 00:00:00 {linuxrc} init
S     0    59     2     0     0 0:0   23:38 00:00:00 [jffs2_gcd_mtd3]
S     0    75     2     0     0 0:0   23:39 00:00:00 [flush-179:0]
S     0    81    53  1448   340 204:18 23:39 00:00:00 /sbin/getty -L ttySGK0 115200 vt100
S     0    91     1  1444   204 0:0   23:39 00:00:00 telnetd
S     0   144     2     0     0 0:0   23:39 00:00:00 [kworker/u:2]
S     0   146     2     0     0 0:0   23:39 00:00:00 [gk_debug]
S     0   147     2     0     0 0:0   23:39 00:00:00 [encoder]
S     0   199     2     0     0 0:0   23:39 00:00:00 [RTW_CMD_THREAD]
S     0   202     1  1388   624 0:0   23:39 00:00:00 wpa_supplicant -Dwext -B -ira0 -c/mnt/mtd/ipc/cfg/wpa_supplicant.conf -P /var/pid_wpa_s
S     0   218     1 69240  2472 0:0   23:39 00:00:03 /mnt/mtd/ipc/app/devctrl
S     0   239     1  130m  7488 0:0   23:39 00:00:11 /mnt/mtd/ipc/app/avencode
S     0   278     1  1576   616 0:0   23:39 00:00:00 /tmp/hwwdg
S     0   280     1 11228  1184 0:0   23:39 00:00:00 /mnt/mtd/ipc/app/sd/sd_detect ok
R     0   281     2     0     0 0:0   23:39 00:00:00 [kworker/0:2]
S     0   340     1  4172  1972 0:0   23:39 00:00:00 /mnt/mtd/ipc/app/AVRecorder
S     0   341     1  3280  1880 0:0   23:39 00:00:00 /mnt/mtd/ipc/app/AVRecSch
S     0   342     1  5312  2384 0:0   23:39 00:00:00 /mnt/mtd/ipc/app/AlarmServer
S     0   397     1 15768  3388 0:0   23:39 00:00:01 /mnt/mtd/ipc/app/rtspd
S     0   423     2     0     0 0:0   23:39 00:00:00 [flush-mtd-unmap]
S     0   462     1  1444   192 0:0   23:39 00:00:00 udhcpc -a -b -i ra0 -x hostname IPC-EWLK-053685-XUBDH -p /var/dhcp.pid
S     0   977     1  1448   168 0:0   23:39 00:00:00 httpd -p 8080 -h /mnt/mmc/sonoff-hack/www/ -c /tmp/httpd.conf
S     0   990     1   884   200 0:0   23:39 00:00:00 pure-ftpd (SERVER)
S     0  1003     1  1060   256 0:0   23:39 00:00:00 dropbear -R
S     0  1025     1  1776   716 0:0   23:40 00:00:00 wsdd --pid_file /var/run/wsdd.pid --if_name ra0 --type tdn:NetworkVideoTransmitter --xaddr http://%s --scope onvif://www.onvif.org/na
S     0  1046     1  1448   256 0:0   23:40 00:00:00 ntpd -p pool.ntp.org
S     0  1378   977  1456   252 0:0   23:41 00:00:00 httpd -p 8080 -h /mnt/mmc/sonoff-hack/www/ -c /tmp/httpd.conf
S     0  1405   977  1456   252 0:0   23:41 00:00:00 httpd -p 8080 -h /mnt/mmc/sonoff-hack/www/ -c /tmp/httpd.conf
S     0  1419   977  1456   252 0:0   23:41 00:00:00 httpd -p 8080 -h /mnt/mmc/sonoff-hack/www/ -c /tmp/httpd.conf
S     0  1430   977  1456   252 0:0   23:41 00:00:00 httpd -p 8080 -h /mnt/mmc/sonoff-hack/www/ -c /tmp/httpd.conf
D     0  1468  1003  1084   476 0:0   23:41 00:00:00 dropbear -R
S     0  1477  1468  1460   436 pts0  23:42 00:00:00 -sh
R     0  1483  1477  1444   356 pts0  23:42 00:00:00 ps -l
[root@GK]# 
`
[image]

alekslyse, 2020-10-23:
strace on the onvif command, plus pictures of the camera info

[image]
[image]
[trace.txt](https://github.com/roleoroleo/sonoff-hack/files/5430634/trace.txt)

alekslyse, 2020-10-23:
And here is trace with no fork

[trace_no_fork.txt](https://github.com/roleoroleo/sonoff-hack/files/5430773/trace_no_fork.txt)

roleoroleo, 2020-10-23:
@alekslyse 
Try this.
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5431469/onvif_srvd.gz)

I removed the threads related to event management.
If doesn't work, please try again with strace using -f option.

alekslyse, 2020-10-24:
Now the onvif starts again, and synology go through its init tests (check capabilities etc), but does not get the rtsp stream and errors out in the profile max. Attached is pcap.

P.S Ip changed so filter in wireshark now is ip.src == 192.168.1.209 or ip.dst == 192.168.1.209

Also did a trace from terminal

[trace_new.txt](https://github.com/roleoroleo/sonoff-hack/files/5432883/trace_new.txt)
[onvif200.pcncap.zip](https://github.com/roleoroleo/sonoff-hack/files/5432884/onvif200.pcncap.zip)

roleoroleo, 2020-10-24:
We will investigate later about event management threads and segfault.
The pcap shows 2 problem:
- CreateProfile and DeleteProfile that the cam doesn't support.
- GetVideoSourceConfigurationOptions and GetVideoEncoderConfigurationOptions actually not implemented.

For the 1st problem I could try to create a fake response (but it's not what we should).
For the 2nd problem I could try to implement them.

I will start with the 2nd step.

alekslyse, 2020-10-24:
yes when you say it I see no reference to the rtsp video files. I did a trace of blue iris too, and that uses the xml file on port 8080 for all its info, so I guess the systems works differently.

If this helps I added a dump from a known camera using the general onvif addon filter: ip.src == 192.168.1.52 or ip.dst == 192.168.1.52

[onvif202.pcncap.zip](https://github.com/roleoroleo/sonoff-hack/files/5432970/onvif202.pcncap.zip)

roleoroleo, 2020-10-24:
Ok, implemented GetVideoSourceConfigurationOptions and GetVideoEncoderConfigurationOptions.
Try this:
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5433172/onvif_srvd.gz)

alekslyse, 2020-10-25:
It starts fine, but crashing when trying to get capabilities. Tried to hardcode the ip and password in the start command too, but no luck

wireshark and strace included

[Archive.zip](https://github.com/roleoroleo/sonoff-hack/files/5434885/Archive.zip)

roleoroleo, 2020-10-28:
Tested with DS214se, it works:
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5454498/onvif_srvd.gz)

alekslyse, 2020-10-28:
Then I have no clue. I set the camera back to default hack/hack - tried in vnc worked. load capabilities work and give video profile. test camera give a check, but no video. clicking ok fails.

Two tcpdumps in the attachment 2002 is clicking all the buttons, load capabilities, video profile and next, 2003 is just clicking video test.

[Archive 2.zip](https://github.com/roleoroleo/sonoff-hack/files/5454732/Archive.2.zip)
[image]

alekslyse, 2020-10-28:
Tried onvif device manager an that cant show live video eather (vlc can) with hack:hack 
[image]

roleoroleo, 2020-10-28:
Probably it's an authentication problem.
Try to set a different user or try to use this.
User: rtsp
Password: 12345678

alekslyse, 2020-10-28:
I will try. Could it be differences on the cameras?

From reset of the camera, what did you do? Setup in ewelink (sound pair) no other changes, popped in a formatted sd with 0.7 of the hack and the latest onvif and put onvif to wifi, reboot.

Anything else you done? It makes no sense it works on one cam and not another

roleoroleo, 2020-10-29:
> From reset of the camera, what did you do? Setup in ewelink (sound pair) no other changes, popped in a formatted sd with 0.7 of the hack and the latest onvif and put onvif to wifi, reboot.

This setup is correct.

Apart from the stream, when you add the camera, does the wizard complete successfully?

alekslyse, 2020-10-31:
No luck with other credentials (that works in vlc). The wizard fails at the screenshot above. Cant resume it.

Im very confused if it really worked in the wizard for you. I get no response for stream during the wizard in any way. Anyone else that can try?

I can add the cam rtsp only of course, but i need ptz

[9 more replies]

closed 2024-05-08 by github-actions[bot]