roleoroleo/sonoff-hack #20: Feature request: cron

duelago, 2020-11-09:
Hello!

Love the firmware. Works great!
Question: I want to use the snapshot feature to grab a frame at a certain time and put together a timelapse video after the winter. I do it right now from a cron script on my computer. Is it possible to put cron on the camera and save the screengrabs directly to the sd-card instead?

Keep up the good work!

 /David

roleoroleo, 2020-11-09:
I did't test it but it should.
If you look at the end of system.sh you can find an example.

```
mkdir -p /var/spool/cron/crontabs/
echo "  0  *  *  *  *  /mnt/mmc/sonoff-hack/script/clean_records.sh $FREE_SPACE" > /var/spool/cron/crontabs/root
/usr/sbin/crond -c /var/spool/cron/crontabs/
```

Pay attention: at the moment, crond is started only if you configure free space check.

duelago, 2020-11-09:
> Pay attention: at the moment, crond is started only if you configure free space check.

Thanks for the quick response! I edited the config script /mnt/mmc/sonoff-hack/script/check_conf.sh with FREE_SPACE=1

Next I rebooted and created the folder and made crontab as in your example.
My snapshot script works fine with cron, but cron is not surviving a reboot. 

The /var/spool/cron/crontab folder is gone.

Any advice? :)

roleoroleo, 2020-11-09:
> Thanks for the quick response! I edited the config script /mnt/mmc/sonoff-hack/script/check_conf.sh with FREE_SPACE=1

You don't need to edit check_conf.sh, simply configure a "SD free space" value in Motion Events page.

> Next I rebooted and created the folder and made crontab as in your example.
> My snapshot script works fine with cron, but cron is not surviving a reboot.

check_conf.sh script check the integrity of the configuration files at boot time.
If you change it nothing happens.

duelago, 2020-11-09:
> You don't need to edit check_conf.sh, simply configure a "SD free space" value in Motion Events page.

Ahhhh. I didn't look at that tab on the web configuration page. I solved it by a ugly hard edit in the config file, but now I know the correct way to do it :) 

Thanks!

closed 2021-05-18 by roleoroleo