roleoroleo/sonoff-hack #21: By default nothing is listening on rtsp port (554).

rkandilarov, 2020-11-24:
Hi!

I've never run the original firmware alone but directly copied https://github.com/roleoroleo/sonoff-hack/releases/tag/0.0.7 on SD and it appears that there is no video stream. 

After login onto the camera I got:
```
[root@GK]# netstat -anp |grep 554
netstat: /proc/net/tcp6: No such file or directory
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
```
Is there any special way to activate the stream? Do I have to pass trough the original firmware for some configuration?

Regards!

EpicLPer, 2020-11-24:
You need to enable it on the Website the camera exposes once you run the Hack.

rkandilarov, 2020-11-24:
I do not see in the Configuration tab nor in any other  tab dedicated option for RTSP. 
`Enable ONVIF Server` is  checked.  
PTZ and snapshots are working.

EpicLPer, 2020-11-24:
Oh, maybe that option is missing then... You probably have to use their app then and enable it there.

roleoroleo, 2020-11-24:
Enable rtsp using sonoff app.

closed 2021-05-18 by roleoroleo