roleoroleo/sonoff-hack #22: Configure WiFi without eWelink

meijerwynand, 2020-11-25:
Greetings, 

Thanks for the work you have done on this project. Your project opens up the hacking possibilities for these devices. 

I prefer to keep my devices "cloud-free" (as far as possible). I was digging around in the os-innards of the camera and I have made some headway in how to configure the WiFi particulars without submitting it via the eWelink app (sound method).

## The short
I have been able to identify a number of changes that allows you to change the wireless credentials without online/app submission. 

### Files involved 
Using `meld` I was able to compare the /mnt/mtd files of a factory reset and wifi-enabled only deployment
- `/mnt/mtd/db/ipcsys.db` - sqlite database containing the WPA stuff
- `/mnt/mtd/ipc/cfg/wpa_supplicant.conf` - WPA stuff
- `/mnt/mtd/ipc/cfg/DyVoiceRecog.bin` - a blank file you need to make 

### Sources
/mnt/mtd/ipc/app/script/dhcp.sh one of the main files that switch between the Wireless (ra0) and Ethernet (eth0)
The command to run (modified) that calls for DHCP on ra0
`udhcpc -v -a -b -i ra0 -x hostname:testing -p /var/dhcp.pid  > /var/dhcp_debug.info` 

### Challenge 
I am able to make the changes to the database, the WPA file and add the `DyVoiceRecog.bin` which _should_ start the WiFi, however this does not happen. 
I can see the connection coming in on my router, the WPA passes and then the device gets in IP, then it stays `unable to route to host`. The DHCP lease does not renew once expired.
Once I connect the LAN cable then I get a DHCP and I can connect to the camera.
I have tried making a number of changes and adding `wpa_cli` commands and the `udhcpc` commands to the `/mnt/mmc/boot.sh` file at various places, but to no avail.

### Almost there
It feels like I am missing something simple and I would like to know if you have any thoughts. I am very sure the config changes I have found and made are close to a victory.

After the WiFi connection has failed and I plug the LAN cable in, when I run the commands from terminal all works 100%, my WPA succeeds, the device gets an IP, I am able to ssh to my root@wan.ip.add.ress, it just fails to boot with WiFi as the only connection method.

## Trail and (t)errors
### Prep for WiFi
- extract sonoff-hack to sdcard (0.0.7)
- load factory defaults of camera if by pressing reset device button
- insert sd card with sonoff hack
- connect lan cable
- boot device
- get files from device
  - get the database file - `/mnt/mtd/db/ipcsys.db`
  - get the wpa_supplicant file -  `/mnt/mtd/ipc/cfg/wpa_supplicant.conf`
- edit the sqlite file and add your wifi details in the 
  - Table: t_sys_param
  - wf_status ->  1
  - wf_ssid -> <your-wifi-ap-ssid>
  - wf_key -> <your-wifi-passkey>
  - wf_auth -> 3
  - wf_enc -> 1

- Edit wpa_supplicant file 
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
network={
	ssid="<the-ssid>"
	scan_ssid=1
	key_mgmt=WPA-EAP WPA-PSK IEEE8021X NONE
	pairwise=TKIP CCMP
	group=CCMP TKIP WEP104 WEP40
	psk="<the-password>"
}
```
- upload the changed files and blank file 
  - `/mnt/mtd/db/ipcsys.db` - sqlite database containing the WPA stuff
  - `/mnt/mtd/ipc/cfg/wpa_supplicant.conf` - WPA stuff
  - `/mnt/mtd/ipc/cfg/DyVoiceRecog.bin` - a blank file you need to make 

### Testing the WiFi changes
Reboot the device (LAN cable stays in)
ssh root@lan.ip.add.ress

```
[root@GK]# wpa_cli -i ra0 status
wpa_state=DISCONNECTED
address=xx:xx:xx:xx:xx:xx

[root@GK]# wpa_cli -i ra0 disconnect
OK

[root@GK]# wpa_cli -i ra0 reconnect
OK

-- give a few seconds --

[root@GK]# wpa_cli -i ra0 status
bssid=xx:xx:xx:xx:xx:xx
freq=0
ssid=<your-ssid-will-show-here>
id=0
mode=station
pairwise_cipher=CCMP
group_cipher=CCMP
key_mgmt=WPA2-PSK
wpa_state=COMPLETED
address=xx:xx:xx:xx:xx:xx

```
The above indicates your WiFi is connected, but you will not have an IP address yet..
```
[root@GK]# udhcpc -a -b -i ra0 -x hostname:justtesting -p /var/dhcp.pid  > /var/dhcp.info
route: SIOCDELRT: No such process

[root@GK]# ip ad
..
3: ra0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 192.168.13.37/24 brd 192.168.13.255 scope global ra0
...
```

Connected!!

As you can see by now, the change in WiFi details are working and you can connect to your Access points

### Sticky part

I just cant seem to get the WiFi to start correctly at boot. Not sure if it is config related, runlevel, or just pebkac 

You inputs and expertise in the matter is highly appreciated and valued. 

## In closing
I trust feature like "Configure your WiFi from the web interface" would be a welcomed addition to the project.

roleoroleo, 2020-11-26:
I did not go into this part very much because I am connected in lan.
Do you know where the cam choose between wireless and cable?
Is there a script or a config?
Or it's automatic and when the cam starts, it tries to connect using lan and then wireless?

meijerwynand, 2020-11-26:
I use `grep` to find whatever I am looking for these are some of the observations.

>  Do you know where the cam choose between wireless and cable?

No, 
- looking for references it appears there is code in `/mnt/mtd/ipc/app/devctrl`, however this appears to be a binary
- best guess is an `if` statement wether the WiFi is configured which will allow / attempt a connection
- `/mnt/mtd/ipc/script/dhcp.sh` is called in `/mnt/mtd/ipc/app/devctrl` and it receives 2 parameters `interface` and `hostname`, looking at the script both is uses `udhcpc` to get the DHCP, however for both for wireless and LAN it seems to use the same pid file when envoked, making me believe it will only try / use one interface

>  Is there a script or a config?

Not too sure, I did see the sqlite file being changed and WiFi credentials being added there, however it also appears it can be override by using native cli commands (shown in previous post). The majority of my fiddling has lead me to think it's mainly scripts. There are 2 script directories: 
- `/mnt/mtd/ipc/app/script/` - likely the factory reset files to restore
- `/mnt/mtd/ipc/script/` - appears to be where many of the scripts run
In here, `/mnt/mtd/ipc/script/dhcp.sh` is the best reference I have found to LAN and WiFi related details, it includes the creation of a standalone AP if WiFi is not configured ( a configured WiFi appears to be indicated by a blank file `/mnt/mtd/ipc/cfg/DyVoiceRecog.bin`)

>  Or it's automatic and when the cam starts, it tries to connect using lan and then wireless?

From the boot it appears to be a single interface. Either LAN or either WiFi, however, once logged in you can run both WiFi and LAN (even ethernet alias on `eth0`)

```
eth0      Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx
          inet addr:192.168.14.19  Bcast:192.168.14.255  Mask:255.255.255.0
...
eth0:0    Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx
          inet addr:192.168.14.55  Bcast:192.168.14.255  Mask:255.255.255.0
...
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
...
ra0       Link encap:Ethernet  HWaddr 68:B9:D3:93:AC:22  
          inet addr:192.168.13.37  Bcast:192.168.13.255  Mask:255.255.255.0
...
```

## Interesting finds...
- You can start the rtsp manually `rtspd` however it has no --help or whatevers.
- Looking into the tool chain, it would appear you _should_ be able to compile your own service e.g. custom rtsp binaries and run them once the device has booted
- killing `/mnt/mtd/ipc/app/devctrl` kills everything, need to reboot
- The `/etc/network/interfaces` file is missing... it is a symlink to a path that does not exist

## Conclusion
- I am still only able to start WiFi manually once I have SSH'd into the camera
- Running the exact same commands at the bottom `/mnt/mmc/sonoff-hack/script/system.sh` does not return the expected result

This is what I have added at the bottom of `/mnt/mmc/sonoff-hack/script/system.sh` 
```
...
## Network hack
## Set static IP for backdoor entrance
/sbin/ifconfig eth0:0 192.168.14.55 up

## kill dhcp server 
pid=$(ps | grep udhcpd | grep -v grep | awk '{print $1}') && kill $pid || echo 'Nothing to kill'

## kill captive portal 
pid=$(ps | grep '/mnt/mtd/ipc/app/captive_server' | grep -v grep | awk '{print $1}') && kill $pid || echo 'Nothing to kill'

## kill ap
pid=$(ps | grep '/gm/bin/hostapd' | grep -v grep | awk '{print $1}') && kill $pid || echo 'Nothing to kill'

## kill dhcp
pid=$(ps | grep 'udhcpc' | grep -v grep | awk '{print $1}') && kill $pid || echo 'Nothing to kill'

## try again
/gm/bin/wpa_supplicant -Dwext -B -ira0 -c/mnt/mtd/ipc/cfg/wpa_supplicant.conf -P /var/pid_wpa_s_debug

killall udhcpd
/gm/bin/wpa_cli -ira0 reconnect
sleep 20
ifconfig eth0 0.0.0.0
/gm/bin/wpa_cli -ira0  status > /var/wpa_debug.info
killall udhcpc
udhcpc -v -a -b -i ra0 -x hostname:IPC-ZZZZ-XXX-TTTTT-WIFI -p /var/dhcp_debug.pid  > /var/dhcp_debug.info

```

meijerwynand, 2020-11-28:
Preliminary success ... kind off..

WiFi has NOT been configured via the eWeLink app. You can boot the device and obtain WiFi access.

When WiFi is not configured, during boot it attempts to go into AP mode, creating a captive portal and its own DHCP server. After this boot completes, forcefully kill participating processes, and interface start `wpa_supplicant` - config in previous post and bring up interface and DHCP client

It takes bout ~1 min to boot and get WiFI connected with an IP

Not working...
- The captive portal (port 80) seems not to conflict with onvif server (also port 80)
- Timezone is not updated 
- The web interface (port 8080) loads the pages, however, 
  - snapshot is not working, running `ps` shows a number for processes spawned
  - slider buttons all in "off" position. 
  - PTZ still functions 

Functional but not production ready code
This file is a stanalone file (`/mnt/mmc/sonoff-hack/script/network.sh`) being called at the bottom of  `/mnt/mmc/sonoff-hack/script/system.sh`
```
#!/bin/ash
## stop and kill 
LOG_FILE=/tmp/wifi_debug
> ${LOG_FILE}

ps >>  ${LOG_FILE}
echo "--------------------------------------------------------------------------">>  ${LOG_FILE}
echo "killing processes and starting wpa_supplicant">>  ${LOG_FILE}
killall captive_server
killall hostapd
killall udhcpd

wpa_supplicant -Dwext -B -ira0 -c/mnt/mtd/ipc/cfg/wpa_supplicant.conf -P /var/pid_wpa_s

echo "--------------------------------------------------------------------------">>  ${LOG_FILE}
ps >>  ${LOG_FILE}
ifconfig ra0 >> ${LOG_FILE} 2>&1 

## disconnect wifi
echo "`date` : wpa_cli -i ra0 disconnect" >> ${LOG_FILE}
wpa_cli -i ra0 disconnect >> ${LOG_FILE} 2>&1

## stop interface
echo "`date` : ifconfig ra0 down" >> ${LOG_FILE}
ifconfig ra0 down >> ${LOG_FILE} 2>&1 

## kill any running based on pid file 
UDHCPC_PID="/var/dhcp.ra0.pid"
if [[ -f "${UDHCPC_PID}" ]]; then 
  echo "`date` : removing pid ${UDHCPC_PID}"  >> ${LOG_FILE}
  kill `cat ${UDHCPC_PID}`
  rm ${UDHCPC_PID}
fi

## kill wifi done... 

## bring up interface 
echo "`date` : ifconfig ra0 up" >> ${LOG_FILE}
ifconfig ra0 up >> ${LOG_FILE} 2>&1
ifconfig ra0 >> ${LOG_FILE} 2>&1 
sleep 3
wpa_cli -i ra0 reconfigure >> ${LOG_FILE} 2>&1

## bring up wifi
echo "`date` : wpa_cli -i ra0 reconnect" >> ${LOG_FILE} 2>&1
wpa_cli -i ra0 reconnect >> ${LOG_FILE} 2>&1

## attempt a dhcp ip
udhcpc -i ra0 -b -S -x hostname:123-123 --tryagain 45 -p ${UDHCPC_PID} >> ${LOG_FILE} 2>&1
```

shafr, 2021-05-26:
Well I pretty much have the same issue - my password is 60+ characters long and i'm not able to set it up using sound prompt. 
I had to use cable, but i'm very happy if I would be able to set it somehow...

kruzer, 2021-10-02:
Another solution i found useful:
1. Temporary change your WIFI password in Access Point to a short-simple one.
2. Pair with eWeLink
3. Boot with sonoff-hack and ssh to camera
4. Change the short password in /mnt/mtd/ipc/cfg/wpa_supplicant.conf to the appropriate one in line:
```
	psk="<Real Password>"
```
5. Update it in db: 
```
/mnt/mmc/sonoff-hack/bin/sqlite3 /mnt/mtd/db/ipcsys.db "update t_sys_param set c_param_value='<Real Password>' where c_param_name='wf_key';"
```
6. Recover your real password in AP and reboot the camera

shafr, 2021-10-05:
Thanks for manual @kruzer .
I created a hotspot wifi on my laptop & followed your instructions. 
I would also add - if you want to switch wifi name - param is `wf_ssid`.
If you want to check param values:
```
/mnt/mmc/sonoff-hack/bin/sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_sys_param" | grep wf
```

jezzaaa, 2022-05-06:
Presumably, the wpa_supplicant.conf file is populated from the database at boot time, so I could edit the sqlite db file only, then reboot.

I'd love for the sonoff-hack code to provide an interface to setting this.

jezzaaa, 2022-05-06:
> Thanks for manual @kruzer . I created a hotspot wifi on my laptop & followed your instructions. I would also add - if you want to switch wifi name - param is `wf_ssid`. If you want to check param [...]

Or, use SQL wildcards instead of grep:

```
/mnt/mmc/sonoff-hack/bin/sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_sys_param where c_param_name like 'wf_%'"
```

roleoroleo, 2022-05-07:
> I'd love for the sonoff-hack code to provide an interface to setting this.

I think I could add it.

roleoroleo, 2022-05-17:
Check this commit:
https://github.com/roleoroleo/sonoff-hack/commit/32c2f55c248ecce3f650d82df51a354d60c55e66

jezzaaa, 2022-05-18:
> Check this commit: [32c2f55](https://github.com/roleoroleo/sonoff-hack/commit/32c2f55c248ecce3f650d82df51a354d60c55e66)

Awesome, thanks @roleoroleo .

I don't know how to apply the commit without manually editing files. So I'll wait until the next version bump, so I can use the "Upgrade Firmware" button. Unless there's a clean way to apply this, that I'm not aware of?

closed 2024-05-08 by github-actions[bot]