roleoroleo/sonoff-hack #24: Suggested hack - zoneinfo, syslog and cron

meijerwynand, 2020-11-27:
Greetings, 

I was unable to get the timestamps on my camera to display according to my timezone (I did a factory reset and did not use the eWeLink app to configure anything). I also wanted to have my cronjobs saved and not lost after each reboot.

## Code stuff
### Script
The code for all three these matters:

```
#!/bin/ash
## 
## set time, start syslog, the start crond

## start using syslog to see what is happening
SONOFF_HACK_PREFIX="/mnt/mmc/sonoff-hack"

SYSLOG_DIR="${SONOFF_HACK_PREFIX}/var/log"
SYSLOG_FILE="${SYSLOG_DIR}/syslog"

## where your crontabs are going to live
CRONDIR="${SONOFF_HACK_PREFIX}/var/spool/cron/crontabs"

## make a copy of your local zoneinfo file (generally /usr/share/zoneinfo),
## rename it and upload it to the location below
ZONEINFO_FILE="${SONOFF_HACK_PREFIX}/etc/localtime"
NTP_SERVER=192.168.13.1
## -----------------------------------------------------------------------------

##  ===== syslog stuff  =====
## use prep to use syslog
if [ ! -d "${SYSLOG_DIR}" ]; then 
  /bin/mkdir -p "${SYSLOG_DIR}"
fi

if [ ! -f "${SYSLOG_FILE}" ]; then 
  /bin/touch "${SYSLOG_FILE}"
fi

## kill any existing 
/usr/bin/killall syslogd &>/dev/null

## start syslog
/sbin/syslogd -O "${SYSLOG_FILE}" -D
## -----------------------------------------------------------------------------

##  ===== zone info stuff =====
## remove current zoneinfo details, replace with your zoneinfo details
if [[ -f "${ZONEINFO_FILE}" ]]; then
  /bin/rm /etc/localtime
  /bin/ln -s "${ZONEINFO_FILE}" /etc/
  /usr/sbin/ntpd -n -q -N -p "${NTP_SERVER}"
  hwclock --systz
fi
## -----------------------------------------------------------------------------

##  ===== cron stuff   =====
## 
if [ ! -d "${CRONDIR}" ]; then 
  /bin/mkdir -p "${CRONDIR}"
fi

## symlink 
## crond seems to ignore the -c flag, so, now it gets created and symlinked
/bin/mkdir -p /var/spool/cron/
ln -s "${CRONDIR}" /var/spool/cron/

## kill any existing 
/usr/bin/killall crond &>/dev/null

## start cron 
/usr/sbin/crond -b -S -l 8 
```

### Installing
- create a new file `/mnt/mmc/sonoff-hack/script/crond.sh` with the contents shown above
- add `/mnt/mmc/sonoff-hack/script/crond.sh` to the bottom of `/mnt/mmc/sonoff-hack/script/system.sh` as a single line

```
...
## Added crontab 
/mnt/mmc/sonoff-hack/script/crond.sh 
```

Disable the ntpdate in the database file - `/mnt/mtd/db/ipcsys.db` (not sure if this really does anything)
`UPDATE "main"."t_sys_param" SET "c_param_value"=0 WHERE c_param_name = "NTPEnable"`
This was done via sqlitebrowser, copy the file locally, change it and then replace the running version on the camera

## Notes
### crontab
After various attempts it appears the `crond` does not honor the  `-c $work_path` flag. For this work around a symlink is created with the actual saving of the crontabs to the `/mnt/mmc/sonoff-hack/` path. Using this method also allows the crontabs to be saved to the SD card and they remain even after reboots.

### syslog
By troubleshooting the initial `crond` issues, `syslogd` as later added to aid in understanding the strange behavior of `crond` (not honoring `-c`). It turned out to actually be a welcomed addition and decided to keep it. This is also save to the SD card which can serve as "off-line" logs.

### zoneinfo
I ended up finding a copy of my zoneinfo on my local pc, so I uploaded it. During boot I remove the default `/etc/localtime` (Hong Kong) and do another symlink to the copy of my zoneinfo file. Followed by a timesync via `ntpd` and updating the `hwclock` to the (new) system zoneinfo.

Now once I have booted the device and logged in, my time is correct
```
[root@GK]# date
Fri Nov 27 16:03:45 SAST 2020

```

### *nix paths 
Here it is attempted to conform to the standard `var/` `etc/` paths. Similar to a *nix environment where stuff generally is found, it is now only prefixed with the SONOFF_HACK path, 
- crons -> ${SONOFF_HACK_PREFIX}/var/spool/cron/crontabs
- syslog -> ${SONOFF_HACK_PREFIX}/var/log/syslog
- localtime ->  ${SONOFF_HACK_PREFIX}/etc/localtime

## Conclusion
These changes seem to keep after reboots and helps with fiddling with the device. I trust someone will find this useful or even better some of these ideas make it into the project.

roleoroleo, 2020-11-27:
About timezone, I already implemented (but not committed) an option to change timezone directly inside sqlite db.
In this manner it's compatible with the app.

I will try to integrate cron and syslog in the hack.

Thank you for your code.

closed 2020-11-27 by meijerwynand

roleoroleo, 2020-12-07:
Syslog is missing, I will check if I can integrate it.