roleoroleo/sonoff-hack #25: PTZ control via Zoneminder

nigelspowage, 2020-11-29:
Has anyone got PTZ controls working through Zoneminder ?

I saw the readme said to use ONVIF camera for Blue Iris so I attempted using the same for Zoneminder but with no luck.

attilabalazsy, 2022-02-21:
I'm playing with the same, no luck yet

attilabalazsy, 2022-02-22:
Yes, ONVIF Device Manager is able to control the PTZ. My issue with zoneminder is, that it is in a different subnet, so it does not finds the camera. This is what I need to solve first, then I can share more info here.

roleoroleo, 2022-02-23:
If the cam is in a different subnet, wsdd discovery will not run successfully.
You need to input the ip address manually.

roleoroleo, 2022-02-24:
Try this:
[image]

attilabalazsy, 2022-03-06:
I've implemented the native PTZ control for zoneminder. If you are interested, I can share the control script.

roleoroleo, 2022-03-06:
Yes, please.

attilabalazsy, 2022-03-06:
Here is it (pls rename it to .pm)
[SonoffHack.pm.txt](https://github.com/roleoroleo/sonoff-hack/files/8193473/SonoffHack.pm.txt)

Installation details are in the description section of the script (at the bottom).
It is working from ZM UI (1.36.12) and also from ZM Ninja mobile app
I was not testing it very detailed and I'm also very new to ZM, so sorry for any mistakes. :)

roleoroleo, 2022-03-07:
Thank you for sharing it.

avg-I, 2022-10-24:
It would be great if @attilabalazsy 's script was integrated into zoneminder.
Have you considered submitted a PR there?

closed 2024-05-08 by github-actions[bot]