roleoroleo/sonoff-hack #28: Version 0.0.7 onvif_srvd segmentation fault

awicenec, 2020-12-20:
First of all: Thanks for this amazing work! I've installed version 0.0.7 without any issues, but then realized that the onvif_srvd daemon did not survive after startup. I then used the exact same settings as in the system.sh script to try and start the daemon with --no_fork and got a segmentation fault. After that tried various things removing some of the additional options, but no luck. Also re-installing did not solve the issue. The only thing that seems to be working is onvif_srvd -h:

 ===============  Help  ===============
 Daemon name:  onvif_srvd
 Daemon  ver:  1.1.0
 Build  mode:  release
 Build  date:  Oct 23 2020
 Build  time:  11:43:15

I also downgraded to 0.0.6 and that worked fine. I then copied the 0.0.6 onvif_srvd binary into the 0.0.7 version and that also works. It also seems that all the other features, not relying on the onvif_srvd of 0.0.7 are working.

Some additional info:

1. I'm running the camera over WIFI
4. Firmware Version | 0.0.7
5. Base Version | V2524.1.245build20191030
6. Model | GK-200MP2B
7. Device ID | a62000169d

Since it took me quite long to even realize that the reason why I was not able to connect using my onvif client was the fact that the server simply was not running I would suggest to add a regular check into the web interface showing that the various daemons are working.

roleoroleo, 2020-12-20:
I know this problem.
When I added event management to onvif_srvd I started to notice segmentation fault.
I tried to solve the problem without success.
So now I restored the old version: https://github.com/roleoroleo/sonoff-hack/commit/f399296a00ff3a2965e959d555fe10888b6e51ea
It is a pity because the implementation of the events was very useful.
But I can't solve the segfault...

Try this binary please: 
[onvif_srvd.gz](https://github.com/roleoroleo/sonoff-hack/files/5720846/onvif_srvd.gz)

urbatecte, 2020-12-20:
I've just test your binary and onvif server is now recognized thanks !
PTZ is KO.

osos, 2020-12-20:
I just tested.

Worked with Home Assistant Onvif integration - also ptz commands

adewilt8, 2020-12-29:
Hi guys, so cool to read that this has been resolved. Can you please explain how to apply this update with the onvif_srvd?
Do I need to downgrade to 0.0.6 first?

roleoroleo, 2020-12-30:
I will release a new version asap.

roleoroleo, 2020-12-30:
Here it is: version 0.0.9.

EasyPP, 2021-01-07:
Works for me

closed 2024-05-08 by github-actions[bot]