roleoroleo/sonoff-hack #30: Receiving motion detected notification through eWeLink app

codebullfrog, 2020-12-30:
After configuring the cam to have cloud disabled I still receive motion detected notifications through the eWeLink app.
When I go into the app my cameras still say offline and can't be accessed otherwise.

roleoroleo, 2020-12-30:
When the cam starts, for a few second, it connects to the cloud, until it's killed by the script.
So it's possibile that you receive messages during the boot phase.

fcsegalla, 2020-12-30:
I receive all the time

Eng. Fernando Segalla
Diretor de Tecnologia
segalla@intralab.com.br
Fone: (55 11) 5555-3296
https://www.intralab.com.br

roleoroleo, 2020-12-30:
Please check if colink process is running.
Post the output of the following commands.

```
ps | grep colink
netstat -anp
```

fcsegalla, 2020-12-30:
I'm out of town, i will check next week

codebullfrog, 2021-01-04:
I also receive these whenever I allow the cam internet access through my router not just on boot. I ran the two commands as requested.

[root@GK]# ps | grep colink
 2031 root      1444 S

[root@GK]# netstat -anp
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN      437/rtspd
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      1159/onvif_srvd
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      1048/httpd
tcp        0      0 0.0.0.0:8081            0.0.0.0:*               LISTEN      437/rtspd
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1106/dropbear
tcp        0      0 0.0.0.0:65530           0.0.0.0:*               LISTEN      437/rtspd
tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN      254/avencode
tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN      254/avencode
tcp        0      0 192.168.0.101:37266     192.168.0.20:1883       ESTABLISHED 1122/mqtt-sonoff
tcp        0    176 192.168.0.101:22        192.168.0.4:52184       ESTABLISHED 1928/dropbear
netstat: /proc/net/tcp6: No such file or directory
udp        0      0 127.0.0.1:11010         0.0.0.0:*                           254/avencode
udp        0      0 127.0.0.1:49938         0.0.0.0:*                           229/devctrl
udp        0      0 127.0.0.1:19000         0.0.0.0:*                           364/AVRecSch
udp        0      0 127.0.0.1:6970          0.0.0.0:*                           437/rtspd
udp        0      0 255.255.255.255:6970    0.0.0.0:*                           437/rtspd
udp        0      0 127.0.0.1:6971          0.0.0.0:*                           437/rtspd
udp        0      0 255.255.255.255:6971    0.0.0.0:*                           437/rtspd
udp        0      0 0.0.0.0:35156           0.0.0.0:*                           229/devctrl
udp        0      0 127.0.0.1:46430         0.0.0.0:*                           365/AlarmServer
udp        0      0 0.0.0.0:17503           0.0.0.0:*                           229/devctrl
udp        0      0 127.0.0.1:54642         0.0.0.0:*                           254/avencode
udp        0      0 0.0.0.0:3702            0.0.0.0:*                           1177/wsdd
udp        0      0 127.0.0.1:45957         0.0.0.0:*                           254/avencode
udp        0      0 127.0.0.1:15000         0.0.0.0:*                           365/AlarmServer
udp        0      0 127.0.0.1:15001         0.0.0.0:*                           365/AlarmServer
udp        0      0 127.0.0.1:15002         0.0.0.0:*                           365/AlarmServer
udp        0      0 127.0.0.1:14000         0.0.0.0:*                           229/devctrl
udp        0      0 0.0.0.0:42692           0.0.0.0:*                           254/avencode
udp        0      0 127.0.0.1:39377         0.0.0.0:*                           254/avencode
udp        0      0 127.0.0.1:34774         0.0.0.0:*                           364/AVRecSch
udp        0      0 127.0.0.1:52956         0.0.0.0:*                           363/AVRecorder
udp        0      0 127.0.0.1:12000         0.0.0.0:*                           363/AVRecorder
udp        0      0 127.0.0.1:40433         0.0.0.0:*                           254/avencode
udp        0      0 0.0.0.0:65010           0.0.0.0:*                           229/devctrl
udp        0      0 127.0.0.1:11000         0.0.0.0:*                           254/avencode
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node PID/Program name    Path
unix  2      [ ACC ]     STREAM     LISTENING        790 254/avencode        /tmp/aaa
unix  2      [ ]         DGRAM                       551 213/wpa_supplicant  /var/run/wpa_supplicant/ra0
unix  3      [ ]         STREAM     CONNECTED       2323 1122/mqtt-sonoff
unix  3      [ ]         STREAM     CONNECTED       2322 1122/mqtt-sonoff

roleoroleo, 2021-01-05:
As you can see in the netstat ouput there are no open sockets.
So, I don't know how the cam can send messages outside your lan.

Anyway, you could try to apply this patch: https://github.com/roleoroleo/sonoff-hack/commit/3c448cf6164744ad14ffd5ef83497828499034f9

codebullfrog, 2021-01-09:
How do I compile the code? I'm not the most familiar with compiling.

codebullfrog, 2021-01-13:
I have installed version 0.0.9 (which I think contains the above patch), but i am still receiving the motion detected notifications on the app when the camera is allowed internet access.

roleoroleo, 2021-01-16:
Please check again connections.
If you find an established tcp connection I could try to blacklist it.

codebullfrog, 2021-01-17:
I looked at the tcp connections again (when the cam is blocked from internet) and found the following connection.

tcp        0      1 192.168.0.111:40376     47.92.0.195:80          SYN_SENT    229/devctrl

After I allowed internet access I found this

tcp        0      0 192.168.0.111:58753     13.52.12.176:8081       TIME_WAIT   -

roleoroleo, 2021-01-17:
I will check them.

roleoroleo, 2021-01-17:
And try to update your system.sh with this new list:
```
    echo "127.0.0.1               eu-dispd.coolkit.cc" >> /etc/hosts
    echo "127.0.0.1               eu-api.coolkit.cn" >> /etc/hosts
    echo "127.0.0.1               testapi.coolkit.cn" >> /etc/hosts
    echo "127.0.0.1               push.iotcare.cn" >> /etc/hosts
    echo "127.0.0.1               www.iotcare.cn" >> /etc/hosts
    echo "127.0.0.1               alive.hapsee.cn" >> /etc/hosts
    echo "127.0.0.1               upgrade.hapsee.cn" >> /etc/hosts
    echo "127.0.0.1               hapseemate.cn" >> /etc/hosts
    echo "127.0.0.1               iotgo.iteadstudio.com" >> /etc/hosts
    echo "127.0.0.1               baidu.com" >> /etc/hosts
    echo "127.0.0.1               sina.com" >> /etc/hosts
```

codebullfrog, 2021-01-18:
The hosts file was as you have posted.

I added the lines to system.sh and have confirmed they were written to the hosts file on start up, however I am still getting the notifications and the connection to 13.52.12.176 still shows up as:

192.168.0.111:53982     13.52.12.176:8081       SYN_SENT    342/AlarmServer

roleoroleo, 2021-01-25:
Try to add a prohibit route:
`ip route add prohibit 13.52.12.176/32`
(after the echo 127.0.0.1 list in the system.sh).

codebullfrog, 2021-01-29:
that seems to have stopped the notifications.

closed 2024-05-07 by github-actions[bot]