roleoroleo/sonoff-hack #31: Prevent repositioning on boot

iiluu, 2021-01-10:
When booted from original firmware, the camera doesn't move after a power failure.
With sonoff-hack, even if I configured a start position, if there is a power failure, the camera will fully turn to one side and then will return to the start position.
Is it possible to remove this useless motion and have it stay put?

(Awesome hack, btw! I especially appreciate the boot silence and the snapshot feature.)

EpicLPer, 2021-01-10:
You can move the camera via PTZ (either via Website or ONVIF) to the position you want and then save that position as the startup one on the Web Interface.

iiluu, 2021-01-10:
> 
> 
> You can move the camera via PTZ (either via Website or ONVIF) to the position you want and then save that position as the startup one on the Web Interface.

Thanks for your answer.

Yes, it goes to the correct position every time. The issue is that after a power failure, even though it is already in the correct position, it starts moving fully to one side and then comes back to the configured startup position. I might be doing something wrong, so tomorrow I'll double check and eventually make a video of it and post the settings.

EpicLPer, 2021-01-10:
This is normal behavior. When the camera starts up it doesn’t know in what position the camera head is in currently. It has to do some calibration steps in order to know where it is again.

iiluu, 2021-01-10:
I get that, but if I remove the microSD and reboot, the original firmware doesn't move the camera at all, so I presumed it has to do with the custom firmware. Is the calibration something that can be turned off?

EpicLPer, 2021-01-10:
The eWeLink app doesn’t care what its position is since it doesn’t actively need to know where the camera head looks at.  
But the custom firmware has the function of precisely going to a pre-defined location where it always needs to know where the head currently is. Since the original app doesn’t have that functionality there’s no need for such a calibration.

roleoroleo, 2021-01-16:
That's what @EpicLPer says is correct.
The tour during the boot is necessary to know the position and the pan/tilt limits.
If I remove this step the preset management can't work.

iiluu, 2021-01-21:
Thanks @EpicLPer @roleoroleo for clearing that out.

closed 2021-01-21 by iiluu

knoffelcut, 2023-05-09:
My apologies for resurrecting this issue, but I believe @iiluu has a valid concern.

Where I live we are currently experiencing rolling blackouts so my camera boots around 4 times per day. The location of the "preset on boot" option is not consistent across reboots. I have never got it to go the the exact location, and sometimes it is off by 30deg.

It might just be a dodgy mounting on my side or the fact that I'm only using a 1A PSU instead of the 1.2A indicated on the packaging.

Regardless, I still have a need to disable the calibration step on boot.

In the meantime I have found that the whole PTZ feature can be disabled by commenting these lines
```
    insmod /mnt/mtd/ipc//app/drive/ptz_drv.ko factory="Links" AutoRun=1 Horizontal=3500 Vertical=900
    if [[ $(get_config PTZ_PRESET_BOOT) != "default" ]] ; then
        (sleep 20 && /mnt/mmc/sonoff-hack/bin/ptz -a go_preset -f $SONOFF_HACK_PREFIX/etc/ptz_presets.conf -n $(get_config PTZ_PRESET_BOOT)) &
    fi
```
in [system.sh](https://github.com/roleoroleo/sonoff-hack/blob/master/src/static/static/sonoff-hack/script/system.sh)

I might experiment some more at a later stage, I expect it should be possible to disable just the calibration on boot and keep the PTZ control through the web interface.

roleoroleo, 2023-05-14:
You probably need to comment out this line as well:
```
rmmod ptz_drv.ko
sleep 1
```