roleoroleo/sonoff-hack #32: Motion events in the ewelink app but not in the webpage

KhurramFHassan, 2021-01-13:
I can see motion events in the ewelink app (through the local storage option) but there no such events listed in the webpage "Motion Events".

roleoroleo, 2021-01-16:
I will check it.

roleoroleo, 2021-01-16:
Does the mp4 file exist?
The event page is fed with the mp4 files found in /tmp/sd/record.
If the event was fired but the file was not created, you can't see it in the event page.

cjonhammar, 2021-01-18:
[root@GK]# cd /tmp/sd/record
-sh: cd: can't cd to /tmp/sd/record
[root@GK]# ls -ltr /tmp/sd/
-rw-r--r--    1 root     root             0 Jan  1  1970 sd.ok

roleoroleo, 2021-01-18:
Sorry:
/mnt/mmc/alarm_record

KhurramFHassan, 2021-01-19:
/mnt/mmc/alarm_record is empty.

roleoroleo, 2021-01-19:
This is my folder:
```
[root@GK]# ls -la /mnt/mmc/alarm_record/
drwxr-xr-x   19 root     root         32768 Jan 18 22:56 .
drwxr-xr-x    8 root     root         32768 Jan 18 23:09 ..
drwxr-xr-x    8 root     root         32768 Dec  6 23:17 20201206
drwxr-xr-x   12 root     root         32768 Dec  7 23:26 20201207
drwxr-xr-x   13 root     root         32768 Dec  8 22:24 20201208
drwxr-xr-x    6 root     root         32768 Dec  9 22:15 20201209
drwxr-xr-x    8 root     root         32768 Dec 10 23:52 20201210
drwxr-xr-x   13 root     root         32768 Dec 11 19:12 20201211
drwxr-xr-x   16 root     root         32768 Dec 12 23:05 20201212
drwxr-xr-x   16 root     root         32768 Dec 13 22:35 20201213
drwxr-xr-x   10 root     root         32768 Dec 14 22:39 20201214
drwxr-xr-x   12 root     root         32768 Dec 15 23:06 20201215
drwxr-xr-x    5 root     root         32768 Dec 16 19:01 20201216
drwxr-xr-x    3 root     root         32768 Dec 17 18:31 20201217
drwxr-xr-x    3 root     root         32768 Dec 20 12:29 20201220
drwxr-xr-x    4 root     root         32768 Dec 28 22:45 20201228
drwxr-xr-x    6 root     root         32768 Dec 30 19:37 20201230
drwxr-xr-x    5 root     root         32768 Jan 18 23:08 20210118
```
Are you able to see the videos inside the app?

KhurramFHassan, 2021-01-19:
The following screenshot is from the eWeLink app:
[image]

The following screenshot is from the ssh session:
[image]

cjonhammar, 2021-01-19:
In my case there is only one video and that is from last reboot of the device. No other motion events has been saved. (the mqtt-events works fine, but no saved video on mmc)

[root@GK]# ls -ltr /mnt/mmc/alarm_record/
drwxr-xr-x    3 root     root          4096 Jan 18 22:12 20210118
[root@GK]# uptime
 10:44:57 up 12:32,  1 users,  load average: 1.49, 1.36, 1.13
[root@GK]# date
Tue Jan 19 10:45:04 CET 2021

roleoroleo, 2021-01-19:
@roleoroleo 
Check if you have the video files inside another folder.
/tmp/sd/*

roleoroleo, 2021-01-19:
And my config is exactly as you described above:
[image]

cjonhammar, 2021-01-19:
Changing that setting didnt help me. After a restart there is only that restart-motion event on the mmc (I have disable cloud enabled). (/mnt/mmc/log/ is also empty)

closed 2024-05-07 by github-actions[bot]