roleoroleo/sonoff-hack #33: PTZ - negative Y axis position

blackdot, 2021-01-18:
Hello, I got a small proble with PTZ on Y axis.
When I set Y axis to "0", it is aiming to -10deg.
`bin/ptz -a go -x 1850 -y 0 -d`
Whne I set to 900, it is aiming 90deg (UP to the ceiling)
`bin/ptz -a go -x 900 -y 0 -d`

But I need to have a look more down (-50deg). Camera is capable to do so. After several down movements I am aiming fully down.
`bin/ptz -a down -t 1000 -d`

When I wanted to save this preset, it will read negative value from Y axis.
`bin/ptz -a set_preset -n 1 -e obyvak -d -f etc/ptz_presets.conf`
`Current position: (1851, -804)`

And issue now, if there is negative value saved in ptz_presets.conf, it will not do any movement.
I am testing this on version 0.0.9

Possible solutions:
1. Is it possible to extend range of Y axis? I would need something like 1700. When it moves fully up, it will set this value as 1700 and then there is a room to move fully down.
2. Reverse Y axis movement. Homing position will be down.

roleoroleo, 2021-01-18:
Strange behavior.
For my cam, 0 and 900 are the limits.
If I try to push it with my hand the cam doesn't move, both in the lower and in the upper limit.
Does your camera take a tour when it starts up?
If yes, does it reach the lower/right limit?
Maybe a different sensor or motor?

blackdot, 2021-01-19:
I tried to use your approach, **set Y=0** and **then by hand push it fully down**. It rotated quite a lot.
After next reboot and re-homing now 0 is really full down.
Need to tear it down and have a look how is the homing achieved. (gyro, endstop, resistance of motor, or just rotating long enough...)

closed 2021-01-19 by blackdot

roleoroleo, 2021-01-19:
Maybe this problem?
https://github.com/roleoroleo/sonoff-hack/issues/5#issuecomment-699478071

If the cam reaches the limit and you try to move it beyond, the position counter increases.
And the cam loses the coordinates.
Maybe you moved the cam beyond the limits?