roleoroleo/sonoff-hack #34: PTZ via onvif?

illivanillie, 2021-02-03:
Hi, great stuff here! Love it!
Is there a way to integrate the PTZ commands or presets via onfiv (into home assistant).
I can't seem to find a was to control the cam. Only via the webpage.
Or is it just me?
Maybe you could help me :)

Thank you for your work!
Greetings from Germany

roleoroleo, 2021-02-03:
Ptz should work with onvif.
Try it with odm.

illivanillie, 2021-02-04:
Well i was trying to get it to work, was playing with the configuration etc., but I'm not able to. 
I don't know what you mean by odm.
In case you are familiar with home assistant: there are only two entities shown, the two streams.
I will try it with blue iris when i get my hands on a windows machine. 

greetings!

roleoroleo, 2021-02-04:
I didn't try with ha, but i tried it with onvif device manager and it works.

urbatecte, 2021-02-04:
Hi,
In HA it does not appear as an entity, but through servic `onvif.ptz`.
You need to give the entty n the servce data:
```
      service: onvif.ptz
      service_data:
        entity_id: camera.sonoff_camera_ch1
        movement: down
```

illivanillie, 2021-02-05:
@urbatecte I just found out, before you commented. Thank you!
@roleoroleo love your stuff! Works great, truly amazing for a cheap cam.

Greetings!

closed 2021-02-05 by illivanillie