roleoroleo/sonoff-hack #44: Webinterface not working, SSH ok (GK-200MP2C?)

joseluislucas, 2021-05-12:
Hi, 
After trying to access the web interface several times I found out that the eWelink app says that my camera is the model GK-200MP2C (the box says GK-200MP2B...) is it possible that this is causing me the problem?.
The hack looks like is partially working as I've done a port scan on the device and says that 21,22,23,80,54,8080 ports are open. I am able to access the camera through SSH (user: root and with no password).  RTSP Streams (av_stream/ch0 and av_stream/ch1) do also work (with huge delay). FTP is also working (user: root, no password needed). 

Hack Version: 0.0.9
Camera model: GK-200MP2C

Many thanks!

roleoroleo, 2021-05-12:
Which port are you using for http?

joseluislucas, 2021-05-13:
> Which port are you using for http?

I'm using 8080

roleoroleo, 2021-05-13:
Please check with netstat, wich daemon is listening on port 8080.
`netstat -anp`

joseluislucas, 2021-05-13:
> Please check with netstat, wich daemon is listening on port 8080.
> `netstat -anp`

Thats the output of netstat -anp

`[root@GK]# netstat -anp
netstat: showing only processes with your user ID
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN      382/rtspd
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      1035/onvif_srvd
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      382/rtspd
tcp        0      0 0.0.0.0:21              0.0.0.0:*               LISTEN      987/pure-ftpd (SERV
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1002/dropbear
tcp        0      0 0.0.0.0:23              0.0.0.0:*               LISTEN      88/telnetd
tcp        0      0 127.0.0.1:6010          0.0.0.0:*               LISTEN      2286/dropbear
tcp        0      0 0.0.0.0:65530           0.0.0.0:*               LISTEN      382/rtspd
tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN      264/avencode
tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN      264/avencode
tcp        0      0 192.168.1.233:22        192.168.1.10:30911      ESTABLISHED 2306/dropbear
tcp        0      0 192.168.1.233:42744     52.57.6.180:443         ESTABLISHED 2325/colink
tcp        0      0 192.168.1.233:8080      192.168.1.10:1043       FIN_WAIT2   -
tcp        0    112 192.168.1.233:22        192.168.1.10:30909      ESTABLISHED 2286/dropbear
tcp        0      0 192.168.1.233:42739     52.57.6.180:443         TIME_WAIT   -
tcp        0      0 192.168.1.233:42737     52.57.6.180:443         TIME_WAIT   -
tcp      466      0 192.168.1.233:8080      192.168.1.10:1047       ESTABLISHED 382/rtspd
tcp        0      0 192.168.1.233:42742     52.57.6.180:443         TIME_WAIT   -
netstat: /proc/net/tcp6: No such file or directory
udp        0      0 127.0.0.1:11010         0.0.0.0:*                           264/avencode
udp        0      0 0.0.0.0:52740           0.0.0.0:*                           306/IOTCare
udp        0      0 127.0.0.1:50203         0.0.0.0:*                           264/avencode
udp        0      0 0.0.0.0:12080           0.0.0.0:*                           306/IOTCare
udp        0      0 127.0.0.1:19000         0.0.0.0:*                           298/AVRecSch
udp        0      0 127.0.0.1:6970          0.0.0.0:*                           382/rtspd
udp        0      0 192.168.1.233:6970      0.0.0.0:*                           382/rtspd
udp        0      0 127.0.0.1:6971          0.0.0.0:*                           382/rtspd
udp        0      0 192.168.1.233:6971      0.0.0.0:*                           382/rtspd
udp        0      0 0.0.0.0:33860           0.0.0.0:*                           306/IOTCare
udp        0      0 127.0.0.1:33353         0.0.0.0:*                           306/IOTCare
udp        0      0 127.0.0.1:40530         0.0.0.0:*                           299/AlarmServer
udp        0      0 0.0.0.0:35156           0.0.0.0:*                           246/devctrl
udp        0      0 127.0.0.1:36184         0.0.0.0:*                           298/AVRecSch
udp        0      0 0.0.0.0:18523           0.0.0.0:*                           306/IOTCare
udp        0      0 0.0.0.0:32108           0.0.0.0:*                           306/IOTCare
udp        0      0 127.0.0.1:41847         0.0.0.0:*                           246/devctrl
udp        0      0 0.0.0.0:60281           0.0.0.0:*                           306/IOTCare
udp        0      0 127.0.0.1:15000         0.0.0.0:*                           299/AlarmServer
udp        0      0 127.0.0.1:15001         0.0.0.0:*                           299/AlarmServer
udp        0      0 127.0.0.1:15002         0.0.0.0:*                           299/AlarmServer
udp        0      0 127.0.0.1:14000         0.0.0.0:*                           246/devctrl
udp        0      0 127.0.0.1:58551         0.0.0.0:*                           264/avencode
udp        0      0 0.0.0.0:37341           0.0.0.0:*                           306/IOTCare
udp        0      0 127.0.0.1:12000         0.0.0.0:*                           297/AVRecorder
udp        0      0 127.0.0.1:53731         0.0.0.0:*                           297/AVRecorder
udp        0      0 127.0.0.1:41708         0.0.0.0:*                           264/avencode
udp        0      0 0.0.0.0:44014           0.0.0.0:*                           306/IOTCare
udp        0      0 0.0.0.0:65010           0.0.0.0:*                           246/devctrl
udp        0      0 127.0.0.1:11000         0.0.0.0:*                           264/avencode
udp        0      0 127.0.0.1:38142         0.0.0.0:*                           264/avencode
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node PID/Program name    Path
unix  2      [ ]         DGRAM                       552 211/wpa_supplicant  /var/run/wpa_supplicant/ra0
unix  2      [ ACC ]     STREAM     LISTENING        856 264/avencode        /tmp/aaa
`

roleoroleo, 2021-05-13:
It seems that your cam expose an rtsp server on port 8080.
`tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      382/rtspd`
Try to change the port of the http daemon to another one.

joseluislucas, 2021-05-13:
> It seems that your cam expose an rtsp server on port 8080.
> `tcp 0 0 0.0.0.0:8080 0.0.0.0:* LISTEN 382/rtspd`
> Try to change the port of the http daemon to another one.

I've done:
'httpd -p 0.0.0.0:8081'

And now there's a new line on netstat that says
'tcp        0      0 0.0.0.0:8081            0.0.0.0:*               LISTEN      30847/httpd'

But when I go to IP:8081 it says Error 404 Not Found

I might have done it wrong.. Any suggestion?

Many thanks :)

roleoroleo, 2021-05-13:
Change the port editing the file system.conf in /mnt/mmc/sonoff-hack/etc

joseluislucas, 2021-05-13:
> Change the port editing the file system.conf in /mnt/mmc/sonoff-hack/etc

Legendary! I am finally able to access the web interface. The only thing that is not working is the PTZ.
- I've tryied pressing the buttons on the PTZ Web Page and nothing happens, some times it whos "PTZ Is not available for that model".
- ONVIF enabled and configured on tinyCam, but PTZ not working.
- And finally, I've tried http://192.168.1.233:8090/cgi-bin/ptz.sh?dir=up/down...  and the camera does not move, but I can see that the web page shows 
`{
}
`

May that be for the port being 8090 (in my case) and not 8080?

Thanks

roleoroleo, 2021-05-13:
The problem is the different model.

roleoroleo, 2021-05-13:
Are you able to make a dump of the flash and send it to me?

joseluislucas, 2021-05-14:
> Are you able to make a dump of the flash and send it to me?

Sure thing, will a copy of /mnt/mtd* be enough?

Here you have it :)
[GK-200MP2C Dump](https://www.mediafire.com/file/gh1qivqj897m69j/GK-200MP2C_dump.zip/file)

Thaks!

roleoroleo, 2021-05-14:
Please try this binary:
[ptz.gz](https://github.com/roleoroleo/sonoff-hack/files/6478712/ptz.gz)

joseluislucas, 2021-05-14:
> Please try this binary:
> [ptz.gz](https://github.com/roleoroleo/sonoff-hack/files/6478712/ptz.gz)

Great! I can use PTZ on tinyCam Pro (Android App) using ONVIF without a problem.

The only thing I noticed is that on the web interface when I press the button to move the camera, the camera does move but most of the time the message `PTZ functions are not available for this camera model.` appears after the first movement. 
Cloud be something timing out? 

You've done an amazing job, sir.

roleoroleo, 2021-05-14:
Please, try this new version.
[GK-200MP2C_0.1.0.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/6481145/GK-200MP2C_0.1.0.tar.gz)

I fixed ptz web page and I changed the default ports for www and onvif (now 80 and 1000).

Could you please check what service is exposed by rtspd on port 8080?
Maybe a mjpg stream?
Try to read it with vlc.

joseluislucas, 2021-05-15:
> Please, try this new version.
> [GK-200MP2C_0.1.0.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/6481145/GK-200MP2C_0.1.0.tar.gz)
>
> I fixed ptz web page and I changed the default ports for www [...]

With the new version, I can access WWW and ONVIF with no problems on stated ports (80 and 1000), but sadly the PTZ is no longer working, not on ONVIF, and not on the web page. When I press the PTZ button it does nothing it doesn't show the "PTZ functions..." like before either.

I've looked for the RTSP stream on 8080 and it looks like the same stream that's on 554 as I can see the stream If I go to `rtsp://IP:8080/av_stream/ch1`  It does not look like MJPG stream cause when I access it through the browser it shows nothing.

Note that the camera is on 50% CPU usage while idle (cloud disabled and streaming RTSP),  but when I go to the webpage It max outs to 100% the whole time. that's the top output.  

> Mem: 32296K used, 4284K free, 0K shrd, 1336K buff, 9644K cached
> CPU: 34.2% usr 50.4% sys  0.0% nic 10.3% idle  0.0% io  0.0% irq  4.9% sirq
> Load average: 11.26 6.48 3.21 2/79 7150
>   PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
>   385     1 root     S    16768 45.6   0 27.1 /mnt/mtd/ipc/app/rtspd
>   264     1 root     S     132m368.7   0 22.3 /mnt/mtd/ipc/app/avencode

Many thanks

roleoroleo, 2021-05-15:
Ok, try this new version, please:
[GK-200MP2C_0.1.0.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/6487515/GK-200MP2C_0.1.0.tar.gz)

joseluislucas, 2021-05-15:
> Ok, try this new version, please:
> [GK-200MP2C_0.1.0.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/6487515/GK-200MP2C_0.1.0.tar.gz)

Great,  I think everything is working fine!

PTZ over ONVIF and webpage is working (web is really slow but I believe it's normal considering the CPU of the device). 

Many thanks!

roleoroleo, 2021-05-15:
0.1.0 released.
Thanks for your support.

attilabalazsy, 2022-02-16:
@joseluislucas, which firmware hack have you applied? The B one? I've exactly the same case you have, the box of my Cam stays it's 200MP2-B, but the eWeLink app stays, the firmware is 200MP2-C. However in my case the firmware is absolutely not loaded, my cam is always loading it's default firmware.  I do not know, what could be the problem.

Thanks!

closed 2024-05-07 by github-actions[bot]