roleoroleo/sonoff-hack #45: PTZ - Set preset not working

skracke, 2021-05-17:
Hi!
I can't get the PTZ go to preset function to work with my custom presets. I have tested to create presets with the name: test on position 0 and with the name Lower-Center on position 1. The creation of the presets seems to work fine but nothing happens when I select one of my own presets and press the go to preset button.

WiFi connected
Firmware Version: 0.1.0
Base Version: V2524.1.245build20191030

Keep up the good work!

/Erik

roleoroleo, 2021-05-17:
Is the problem new with 0.1.0?

skracke, 2021-05-17:
I have not tested any of the previous versions so I don't know. If I use any of the predefined presets (10-14) everything works fine.
/Erik

rcsmota, 2023-04-18:
I'm using firmware 0.1.5 and Base Version | V5520.2053.0504build20230103 and have same problem. 

Predefined presets (10-14) works fine but set preset (0-9) no work, on GUI or ONVIF.

Is there anything that can be done even if it's on the command line?

Thanks

roleoroleo, 2023-04-18:
Please, show me your ptz_presets.conf
`[root@sonoff-hack]# cat /mnt/mmc/sonoff-hack/etc/ptz_presets.conf`

rcsmota, 2023-04-18:
cat /mnt/mmc/sonoff-hack/etc/ptz_presets.conf
0=Entrada|1227|-573
1=Sofa|2239|-573
2=empty|-1|-1
3=empty|-1|-1
4=empty|-1|-1
5=empty|-1|-1
6=empty|-1|-1
7=empty|-1|-1
8=empty|-1|-1
9=empty|-1|-1

Thanks

roleoroleo, 2023-04-18:
Negative number is not ok and will not work.
But I don't know why there is a negative number.
You should try to set again the preset.
The command:
`ptz -a get_coord`
read the actual position.
It should be
0 <= x <= 3500
and
0 <= y <= 1350

rcsmota, 2023-04-18:
Thanks @roleoroleo 

The command also returns a negative value. Is it because it is pointed downwards?

[root@root]# ptz -a get_coord
Current position: (2257, -1519)

It's a photo of the camera positioning.

[image]

roleoroleo, 2023-04-18:
0 should be cam face down and 3500 should be cam face up.
I don't know why it's negative.
Maybe a different position sensor in your model.
Does the cam rotate up/down/left/right when you restart it?

rcsmota, 2023-04-19:
Thanks @roleoroleo 

I did the following, turned off the camera, removed the SD card and boot with the original firmware, the camera rotate up/down/left/right at the beginning, then turned it off and went up again with the SD card, it also rotate up/down/left/right and now everything is working fine.

Thank you very much for the help, I wouldn't go there without help.

roleoroleo, 2023-04-20:
I think it's a sync problem.
When the cam reaches the extreme position the sensor reaches the boundary value (example 3500).
But if you continue to move the cam in the same direction, the cam can no longer move but the value of the position sensor continues to increase.

closed 2024-05-07 by github-actions[bot]