roleoroleo/sonoff-hack #48: arm-sonoff-linux-uclibcgnueabi-gcc

shafr, 2021-05-25:
Where can I find `arm-sonoff-linux-uclibcgnueabi-gcc` package (UBUNTU)? 
I cannot build firmware otherwise, due to error:

```
/bin/sh: 1: arm-sonoff-linux-uclibcgnueabi-gcc: not found
or 
/sonoff-hack/src/busybox/busybox/scripts/gcc-version.sh: line 11: arm-sonoff-linux-uclibcgnueabi-gcc: command not found
```

google does not help much.

roleoroleo, 2021-05-26:
Check the wiki:
https://github.com/roleoroleo/sonoff-hack/wiki/Toolchain

shafr, 2021-05-26:
OK, thank you!
I thought about adding github action that would build artifact automatically after push. 
Does it makes sense ?

roleoroleo, 2021-05-26:
Yes.
crdnl had already done something similar on the Allwinner platform:
https://github.com/roleoroleo/yi-hack-Allwinner/actions

shafr, 2021-06-09:
What would be perfect if you can set up docker container with tooling inside that can be used for build. 
This way github action can be run inside that container & everyone who can build it locally can use that container.

shafr, 2021-06-18:
Ok, one more:
```
./compile.www: 29: yui-compressor: not found
```

roleoroleo, 2021-06-18:
You need to install the package yui-compressor

shafr, 2021-06-18:
Ok, great.
What is output artifact ? Tar'ed contents of build folder ?

roleoroleo, 2021-06-18:
The build folder with some changes.
https://github.com/roleoroleo/sonoff-hack/blob/master/scripts/pack_fw.sh

shafr, 2021-06-22:
Implemented build in #51

closed 2021-06-22 by shafr