roleoroleo/sonoff-hack #49: PTZ Web interface control not working

YanCybulski, 2021-06-14:
Using the latest version (0.1.0) with a fresh sonoff
Clicking the arrows does not do anything

roleoroleo, 2021-06-14:
Which sonoff version?

the-cobbler, 2021-06-14:
Firmware Version | 0.1.0
-- | --
Base Version | V2524.1.276build20201103
Model | GK-200MP2B

Now I see that the firmware didn't upgraded when I clicked it on the android app...not sure why it does not upgrade

roleoroleo, 2021-06-14:
Actually your build is not tested.
I will check it.

roleoroleo, 2021-06-14:
My cam didn't receive the update.
It's at V2524.1.245build20191030

the-cobbler, 2021-06-14:
[image]
It shows that the latest version is lower then the one I have, that's why it didn't "upgrade"

Its a brand new cam

roleoroleo, 2021-06-14:
You probably have a fw version installed in factory and not released as over the air update.
Could you please create a backup?

the-cobbler, 2021-06-14:
Sure, do you have a bash for that?

roleoroleo, 2021-06-14:
```
dd if=/dev/mtd2 of=/mnt/mmc/mtd2.bin
dd if=/dev/mtd3 of=/mnt/mmc/mtd3.bin
dd if=/dev/mtd4 of=/mnt/mmc/mtd4.bin
```

the-cobbler, 2021-06-14:
Already did :)
5 blocks
uploading

the-cobbler, 2021-06-14:
Let me know if you need anything else

roleoroleo, 2021-06-15:
Try this binary:
[ptz.gz](https://github.com/roleoroleo/sonoff-hack/files/6653407/ptz.gz)

the-cobbler, 2021-06-15:
Works.
Thanks!

closed 2024-05-07 by github-actions[bot]