roleoroleo/sonoff-hack #54: Anyone had any luck linking to hikvision?

pme1, 2021-06-26:
Hi
tried loads of different setting to get the sonoff to talk to hikvision but so far no luck, anybody had success?
rgds
chris

Awethentique, 2021-06-26:
Do you have the same ports for ONVIF in the config web page and the Hikvision IP cam setup page?

pme1, 2021-06-28:
Yes, have tried every combination but still wont connect

Awethentique, 2021-06-28:
Looking at the hikvision docs for newer devices here:
https://www.hikvision.com/content/dam/hikvision/en/support/how-to/how-to-document/nvr/How-to-Add-IP-Camera-with-Custom-Protocol.pdf

We are meant to set a custom protocol which would provide the URL

eg.`rtsp://192.168.0.158/av_stream/ch0`

Where hikvision NVR are looking at 
these stream paths instead:

Main stream : streaming/channels/101 or ch1/main/av_stream ;
Sub stream : streaming/channels/102 or ch1/sub/av_stream .

What might help is of we could somehow find a way to configure the sonoff cam to output RTSP to a path matching what Hikvision is expecting

zumlin, 2021-06-30:
I have mine working for a couple for months.

For the IP, put in the IP only, do not put anything else such as "av_stream".

Protocol: ONVIF
Port: 80
Channel: 1
Transfer protocol: TCP

Give it a try.

roleoroleo, 2021-07-05:
I can confirm the post above.
Don't add a path to the ip address.
Use TCP port 1000 (if you are using default settings).

The cam sends to the onvif client the correct rtsp address with the correct path.
rtsp://192.168.0.158/av_stream/ch0
You don't need to specify it.

pme1, 2021-07-08:
The HiWatch (lower end product) doesnt seem to like it all. Time to ditch Hikvision for something else.
Thanks anyway

roleoroleo, 2021-07-08:
Sorry but ONVIF is a non-standard standard.
:(

Awethentique, 2021-07-14:
@roleoroleo I have had some luck.
It turns out i needed to set the following ONVIF settings in the camera web page:
[image]

In my case `ONVIF Profile`  was not set to `both`.
Also make sure the password for RTSP is correct, I had to set from the ewelink app, as the webpage was not updating and kept reverting to an old password. I checked and confirmed this by inspect the password field and change the input type="text" instead of "password"

Therefore I am assuming the stream was not being broadcast for the DVR to pick it up

roleoroleo, 2021-07-16:
> Also make sure the password for RTSP is correct, I had to set from the ewelink app, as the webpage was not updating and kept reverting to an old password. I checked and confirmed this by inspect the password field and change the input type="text" instead of "password"

From webpage, did you change only password or both username and password?

Awethentique, 2021-07-17:
I just updated the password and it worked. Hikvision DVR for some reason throws "unknown error" instead of "password incorrect" or anything about "authentication failure" keeps changing the management port to 554 but thats ok, it still works fine.

closed 2024-05-06 by github-actions[bot]