roleoroleo/sonoff-hack #55: RTSP fails after a few minutes

manJerky, 2021-06-28:
|   |   |
|---|---|
|Firmware Version| 0.1.0 |
|Model| GK-200MP2B |
|Base Version|V2524.1.245build20191030|
|Hack Version| 0.1.0 |

The device is connected via ethernet, but I've fully blocked it from accessing the WAN through my router.
I've disabled cloud access and set the ntp server to my router.
Swap File is enabled.

After a few minutes (<10) of being turned on, the RSTP streams and the snapshot become unresponsive. I would watch them function normally on VLC and then lose connection and not be able to reconnect until I reboot the device.

I'm not sure how to debug or where logs are located but I can upload them if someone tells me where they are. 

Thanks

roleoroleo, 2021-07-05:
If you open the internet access (just for this test) does the cam work?
If you don't call a snapshot but you read rtsp stream does the cam work?

manJerky, 2021-07-06:
I'm unable to allow internet access on my network right now but I have been using a hacky solution.

So it doesn't matter if I view the rtsp or snapshot, both fail in less than 10 minutes. When I manually restart the avencode it will stay up for a few hours, maybe 4 maybe 8 or so, but it will go down again.

My hacky solution is to create this file:

`/mnt/mmc/sonoff-hack/script/restart_avencode.sh`
``` 
#!/bin/ash
is_running=$(ps | grep '/mnt/mtd/ipc/app/a[v]encode' | awk '{print $1}');

echo $is_running;
if [ -z $is_running ]; then
    /mnt/mtd/ipc/app/avencode
fi
```

After creating this file I add a crontab input in the web ui.
`* * * * * /mnt/mmc/sonoff-hack/script/restart_avencode.sh`

Sure it could take up to a minute to restart the service, but realistically the service only stays down a max of 2 or 3 minutes a day now which is great for me.

I wanted to use `/etc/inittab` to auto restart the service there, but it seems I can't change the inittab functionality after boot.

roleoroleo, 2021-07-06:
I will add a watchdog to control avencode like in the yi cams.

roleoroleo, 2021-07-07:
https://github.com/roleoroleo/sonoff-hack/commit/7196b2c9b3d4e4e490aed5e837034830678dc7d9

closed 2024-05-06 by github-actions[bot]