roleoroleo/sonoff-hack #61: Camera port open unable to connect to web interface or SSH

willbrim, 2021-07-27:
Hi all I extracted the firmware using 7zip directly to the SD card, rebooted the camera with the card inserted, ran a port scan on the camera's IP address and see it is listening only on port 23 and 8080. I can not ssh or access the web interface. Has anyone else had this issue?

willbrim, 2021-07-27:
I am trying to access the camera at its ip address via http://192.168.1.32:8080/

willbrim, 2021-08-05:
Thanks I saw that it was updated to port 80 in the notes however,  I can not reach it on 80 and a port scan shows its listening on 8089 and 23 only. I am sure I grabbed the latest release also.

willbrim, 2021-08-05:
[image]

willbrim, 2021-08-05:
Here is an example from today running a test net con on the camera ip for all 3 ports. The camera shows up as unavailable in the ewelink app but I still get motion notifications somehow. I also did not set up wifi on it. I just have it plugged in directly via ethereal. Could that be an issue?

roleoroleo, 2021-08-05:
Probably the hack is not running.
Check the sd card.

willbrim, 2021-08-24:
Thanks for all the help I am not sure what I must be doing wrong here. I have the model B camera and downloaded the B version from the releases section 
[image]
then sed 7-zip and selected extract files to 
[image]
(E is the root letter of the SD card in my case).
Then it shows up as a tar file in the SD card:
[image]
then plugged the camera back in and probed the IP address. Same results. It is listening on 8080 not 80 and not reachable via web at its ip address via http

willbrim, 2021-08-25:
I was able to get the hack loaded! Thanks @roleoroleo !

closed 2021-08-25 by willbrim