roleoroleo/sonoff-hack #62: changing password changes only http password, but not rtsp

avg-I, 2021-08-25:
First, I could not get RTSP working at all until I created "RTSP Link" via eWeLink.
I used a default password there.

Later, I used sonoff-hack web interface to disable the cloud.
I also changed password.
Now the http access requires the new password.
But RTSP still requires the old password created via eWeLink.
The new password results in 401 Unauthorized.

Sonoff-hack version is 0.1.0.
Base firmware on this camera is V2524.1.276build20201103.
It's the same as in issue #49 and I had to use the binary from that issue to get PTZ working.

So, I guess that there can be more incompatibilities between sonoff-hack and that version of the base firmware .

roleoroleo, 2021-08-25:
> I also changed password.
> Now the http access requires the new password.
> But RTSP still requires the old password created via eWeLink.
> The new password results in 401 Unauthorized.

The RTSP service should accept both password.

> Sonoff-hack version is 0.1.0.
> Base firmware on this camera is V2524.1.276build20201103.
> It's the same as in issue #49 and I had to use the binary from that issue to get PTZ working.

I don't remember if I commited the code form https://github.com/roleoroleo/sonoff-hack/issues/49
I will check it.

willbrim, 2021-08-25:
I am having the same symptoms. I was able to get the hack loaded but rtsp does not work at all until creating the ewelink link. In my case however, the rtsp stream dies after about one minute. Only rebooting the camera will restore it. I will post my build tomorrow for reference.

roleoroleo, 2021-08-25:
It's normal that the RTSP stream is not available until you enable it in the app.
I will add this note to the readme.

avg-I, 2021-08-25:
> The RTSP service should accept both passwords.

Unfortunately it accepts only the password that was originally set via the app.
The new password is rejected.

avg-I, 2021-08-25:
User is "rtsp". This user is configured in both the official app and in the sonoff-hack web interface.
Should I try using a different user for sonoff-hack?
Let me do that.

avg-I, 2021-08-25:
Yes, selecting a different user allows RTSP access with both old-user:old-password and new-user:new-password.

avg-I, 2021-08-25:
Although, I am _not_ actually thrilled that old-user:old-password is still allowed access...

avg-I, 2021-09-18:
So 3 weeks down the road everything is still fine.
The camera has been rebooted several times including graceful reboots and unexpected power-offs.

closed 2024-05-06 by github-actions[bot]