roleoroleo/sonoff-hack #66: Motion events are not recorded

kruzer, 2021-09-27:
I've managed to setup two devices with versions 0.1.0 i 0.1.1 of this nice hack. Both cameras hasn't been used/paired in ewlink cloud due to broken sound pairing, but almost everything is working fine with HA and ONVIF integration.
But regardless of any settings in "Motion Events" part of the web config, it never recorder any video file.
How can I debug this problem, what process is responsible for motion events recording?

roleoroleo, 2021-09-28:
The original firmware is responsible for detect the motion and write a new line into t_alarm_log table.
Then, my mqtt-sonoff application, reads continuously the table and checks for new detections.

kruzer, 2021-09-28:
Ok, I see. Maybe some components didn't start? All running processes on my device:
```sh
  PID USER       VSZ STAT COMMAND
    1 root      1448 S    {squashfs_init} /gm/bin/busybox ash /squashfs_init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW<  [khelper]
    7 root         0 SW   [sync_supers]
    8 root         0 SW   [bdi-default]
    9 root         0 SW<  [kblockd]
   10 root         0 SW   [khubd]
   12 root         0 SW<  [cfg80211]
   14 root         0 SW   [kswapd0]
   15 root         0 SWN  [ksmd]
   16 root         0 SW   [fsnotify_mark]
   17 root         0 SW<  [crypto]
   24 root         0 SW   [mtdblock0]
   25 root         0 SW   [mtdblock1]
   26 root         0 SW   [mtdblock2]
   27 root         0 SW   [mtdblock3]
   28 root         0 SW   [mtdblock4]
   29 root         0 SW   [mtdblock5]
   30 root         0 SW   [kworker/u:2]
   33 root         0 SW   [mmcqd/0]
   34 root         0 SW<  [deferwq]
   54 root      1448 S    {linuxrc} init
   60 root         0 SWN  [jffs2_gcd_mtd3]
   82 root      1448 S    /sbin/getty -L ttySGK0 115200 vt100
  163 root         0 SW   [gk_debug]
  164 root         0 SW   [encoder]
  216 root         0 SW   [RTW_CMD_THREAD]
  235 root     69240 S    /mnt/mtd/ipc/app/devctrl
  258 root      130m S    /mnt/mtd/ipc/app/avencode
  300 root      1576 S    /tmp/hwwdg
  305 root     11228 S    /mnt/mtd/ipc/app/sd/sd_detect ok
  329 root      4172 S    /mnt/mtd/ipc/app/AVRecorder
  330 root      3280 S    /mnt/mtd/ipc/app/AVRecSch
  331 root      5312 S    /mnt/mtd/ipc/app/AlarmServer
  412 root         0 SW   [kworker/0:2]
  918 root      1388 S    wpa_supplicant -Dwext -B -ira0 -c/mnt/mtd/ipc/cfg/wpa_supplicant.conf -P /var/pid_wpa_s
 1000 root      1444 S    udhcpc -i ra0 -b -S -x hostname camerasonoff2 --tryagain 45 -p /var/dhcp.ra0.pid
 1030 root      1448 S    httpd -p 80 -h /mnt/mmc/sonoff-hack/www/ -c /tmp/httpd.conf
 1057 root      1060 S    dropbear -R
 1066 root     10188 S    /mnt/mmc/sonoff-hack/bin/mqtt-sonoff
 1078 root     19812 S    /mnt/mtd/ipc/app/rtspd
 1089 root      6888 S    onvif_srvd --conf_file /tmp/onvif_srvd.conf
 1095 root      1776 S    wsdd --pid_file /var/run/wsdd.pid --if_name ra0 --type tdn:NetworkVideoTransmitter --xaddr http://%s:1000
 1096 root      1448 S    {wd_rtsp.sh} /bin/sh /mnt/mmc/sonoff-hack/script/wd_rtsp.sh
 1113 root      1448 S    /usr/sbin/crond -c /var/spool/cron/crontabs/
 1137 root      1448 S    ntpd -p pool.ntp.org
```
Or maybe there is a related setting in the db, alarms are triggered properly (there are records in t_alarm_log) but the table t_alarm_action looks like it is not configured yet:
```sqlite
sqlite> select * from t_alarm_action;
c_index|c_name|c_enable|c_param1|c_param2|c_param3|c_param4
0|store picture to sd card|0||||
1|store video sd card|0||||
2|ftp picture to server|0||||
3|ftp video to server|0||||
4|drive io output port |0|5|||
5|send mail with picture|0||||
6|goto preset|0|1|||
7|Alarm Beep|0|5|||
8|push msg to alarm server|0||||
```

roleoroleo, 2021-09-28:
Check if motion detection is enabled.

YES
```
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_mdarea;"
0|0|0|1920|1080|25|1|P2P_SET
1|928|32|1248|352|50|0|MD1
2|32|368|352|688|50|0|MD2
3|928|368|1248|688|50|0|MD3
```
NO
```
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_mdarea;"
0|0|0|1280|720|70|0|MD0
1|928|32|1248|352|50|0|MD1
2|32|368|352|688|50|0|MD2
3|928|368|1248|688|50|0|MD3
```

kruzer, 2021-09-28:
Looks like it is enabled:
```sh
[root@GK]# ./bin/sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_mdarea;"
0|0|0|1920|1080|25|1|P2P_SET
1|928|32|1248|352|50|0|MD1
2|32|368|352|688|50|0|MD2
3|928|368|1248|688|50|0|MD3
```

roleoroleo, 2021-09-29:
Can you see mp4 files in the /mnt/mmc/alarm_record folder?

kruzer, 2021-10-02:
no, there are no files there

roleoroleo, 2021-10-02:
Did you enable motion detection in camera settings?

kruzer, 2021-10-02:
Camera settings wasn't available in menu, however i could open url http://camerasonoff1/index.html?page=camera_settings, but was unable to save changes.
Finally i solved it by changing wifi_key to an easy-short one, then followed the audio paring procedure. Than I was able to enable recording in ewlink, now recording in sonoff-hack is working ok.

closed 2021-10-02 by kruzer

roleoroleo, 2021-10-02:
> Camera settings wasn't available in menu, however i could open url http://camerasonoff1/index.html?page=camera_settings, but was unable to save changes.

You are right.
I added it in the last version.