roleoroleo/sonoff-hack #67: Allow set preset via URL

pops106, 2021-09-28:
Is there a way I can set a PTZ location preset via a curl command.

For example if one of my other cameras detects motion, front door for example I can set an automation to send a curl command to the sonoff to set it to a certain preset to watch the path down the side of the house.

roleoroleo, 2021-09-28:
You can set a preset (save it in the config file) with this syntax:
`cgi-bin/preset.sh?action=set_preset&num=1&name=Path`

You can go to a preset with this syntax
`cgi-bin/preset.sh?action=go_preset&num=1`

Check the wiki.

bennyboom38, 2023-12-16:
Thanks for the info but when I try with curl command , everything seems to work but the camera does not change his position  to the preset requested.
Here is the log : 

command used : curl -v --user "mylogin:mypassword" http://192.168.X.XXX/cgi-bin/preset.sh?action=go_preset&num=0

`
[1] 3335303
*   Trying 192.168.X.XXX:80...
* Connected to 192.168.X.XXX (192.168.X.XXX) port 80 (#0)
* Server auth using Basic with user 'mylogin'
> GET /cgi-bin/preset.sh?action=go_preset HTTP/1.1
> Host: 192.168.X.XXX
> Authorization: Basic XXXXXXXXXX
> User-Agent: curl/7.85.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Content-type: application/json
< 
{
"error":"true"
* Closing connection 0
}

`

roleoroleo, 2024-01-14:
Did you try it with a browser?
Does it work without auth?

bennyboom38, 2024-01-15:
When I try with a browser I need to authenticate with basic auth login/password and then I get an error page too with :
`{
  "error": "false"
}`

I have not tried without authentication, I'm going to try to disable it and try again

roleoroleo, 2024-01-15:
> { "error": "false" }

Sorry for the misunderstanding but this output means "no errors".
You can remove the authentication in the webgui:
`http://IP-CAM/?page=configurations`

bennyboom38, 2024-01-16:
("Sorry for the misunderstanding but this output means "no errors"." => got it ! )
In fact the field are not filled but when I try to connect with browser I get a popup to insert login password.
I have tried to put a new login/password then to empty fill and save configuration and reboot camera but nothing has changed , when I enter curl command in console a get error and camera does not move.
Maybe there's something wrong with credentials :/, leave fields empty does not seem to remove authentification

roleoroleo, 2024-01-16:
Try to clear the cache of the browser.
If it doesn't work, enter with a ssh client and check the config file
`sonoff-hack/etc/system.conf`

roleoroleo, 2024-01-16:
Try double quote around the url.

BenjaminFalque, 2024-05-11:
Hello ! 
I have just tried with double quote and it's working fine!

` curl -v "http://192.168.x.xxx/cgi-bin/ptz.sh?dir=right"
* processing: http://192.168.x.xxx/cgi-bin/ptz.sh?dir=right
*   Trying 192.168.x.xxx:80...
* Connected to 192.168.x.xxx (192.168.x.xxx) port 80
> GET /cgi-bin/ptz.sh?dir=right HTTP/1.1
> Host: 192.168.x.xxx
> User-Agent: curl/8.2.1
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Content-type: application/json
< 
{
"error":"false"
* Closing connection
`

Thanks a lot for your help !

closed 2024-09-09 by github-actions[bot]