roleoroleo/sonoff-hack #68: Problem with FTP upload and Ewlink view

wlenbin, 2021-10-09:
Hi! Many thanks for your work! Unfortunately after several try, my camera does not upload any videos to FTP server. Server, username and ftp pass is correct (I try use it with Filezilla it's normal working). I see all records in Sonoff hack, but no in my FTP. Do you have any ideas to fix it? 
Also, after sonoff hack was be installed i can not see image from camera in ewlink. Motion akert is normal working in ewlink app, but can not connecting to the camera. Is it possible to use sonoff hack and have normal view in ewlink app?
Million thanks for the answers!

roleoroleo, 2021-10-10:
> Hi! Many thanks for your work! Unfortunately after several try, my camera does not upload any videos to FTP server. Server, username and ftp pass is correct (I try use it with Filezilla it's normal working). I see all records in Sonoff hack, but no in my FTP. Do you have any ideas to fix it?

Try to run the script manually and check the errors.
```
[root@GK]# . /mnt/mmc/sonoff-hack/script/env.sh
[root@GK]# /mnt/mmc/sonoff-hack/script/ftppush.sh start
```
 
> Hi! Many thanks for your work! Unfortunately after several try, my camera does not upload any videos to FTP server. Server, username and ftp pass is correct (I try use it with Filezilla it's normal [...]

You can use both sonoff-hack and ewelink. Simply don't enable "Disable cloud" in configuration web page.

wlenbin, 2021-10-20:
> Try to run the script manually and check the errors.
>
> ```
> [root@GK]# . /mnt/mmc/sonoff-hack/script/env.sh
> [root@GK]# /mnt/mmc/sonoff-hack/script/ftppush.sh start
> ```
>
>
> You can use both sonoff-hack [...]

Hi! 
You can see my screenshot of this script working. Do you have any ideas how to fix FTP upload? 
Thank you!

closed 2024-05-06 by github-actions[bot]