roleoroleo/sonoff-hack #71: Firmware -  27620201103

piechade, 2021-11-24:
Is the version 27620201103 working with this hack?

roleoroleo, 2021-11-24:
Check this long thread:
https://github.com/roleoroleo/sonoff-hack/issues/16

IMHO the onvif implementation by Synology is not standard.
I successfully connected a DS214e with this cam but other users found problems.

piechade, 2021-11-27:
What can I provide you to find the issue together?

roleoroleo, 2021-11-27:
You should capture the traffic between the cam and the nas.
tcpdump for example.
If you are able to capture it inside the nas it's probably better.
Otherwise you can download tcpdump and run tcpdump in your cam (check this repo).
Run it with this command line:
`tcpdump -i wlan0 -w /tmp/sd/capture.pcap`

piechade, 2021-11-27:
It's working, but I have no idea why and what was wrong before.

closed 2021-11-27 by piechade