roleoroleo/sonoff-hack #73: Turning off IR LEDS

vadimbz, 2021-12-06:
Would it be possible to implement subj? I'm using the through glass, and not being able to get rid of IR defeats the purpose.

roleoroleo, 2021-12-07:
There are 3 params that we could test.
```
[root@GK]# . /mnt/mmc/sonoff-hack/script/env.sh
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="InfraredLamp";'
|InfraredLamp|2
```
Maybe we could try 0 or 1...

```
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_para
m_name="LED1Mode";'
|LED1Mode|1
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_para
m_name="LED2Mode";'
|LED2Mode|1
```

roleoroleo, 2021-12-07:
Another:
```
[root@GK]# . /mnt/mmc/sonoff-hack/script/env.sh
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="IRCUTMode";'
|IRCUTMode|2
```

vadimbz, 2021-12-07:
Camera has to be rebooted after each parameter change, right?

vadimbz, 2021-12-07:
If possible, I would like to keep monochrome night mode, and turn off just IR

roleoroleo, 2021-12-07:
> Camera has to be rebooted after each parameter change, right?

I don't know but I think it has to be rebooted.

vadimbz, 2021-12-07:
I'm sorry,  my knowledge of SSH is cursory, how exactly do I edit these? I'm getting the same output as you've shown, but I can't change values at any point.

roleoroleo, 2021-12-07:
Ok, it's a sql client.
There is a db with a table "t_sys_param" that contains all settings.
With SELECT and UPDATE you can change a row in the table.
For example:
```
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=0 where c_param_name="InfraredLamp";'
```
set the value of the parameter InfraredLamp to 0.
You should try different values.
At the moment I tried the following values:
- 0 IR on with light or dark
- 1 IR off with light and on with dark
- 2 same as 1

vadimbz, 2021-12-07:
```
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="LED1Mode";'
|LED1Mode|0
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="LED2Mode";'
|LED2Mode|0
```
No dice, IR still on
```
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="LED1Mode";'
|LED1Mode|2
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="LED2Mode";'
|LED2Mode|2
```
Ditto - IR still on
I'll try other parameters

vadimbz, 2021-12-07:
```
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="LED1Mode";'
|LED1Mode|0
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="LED2Mode";'
|LED2Mode|0
```
IR still on
```
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="LED1Mode";'
|LED1Mode|2
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="LED2Mode";'
|LED2Mode|2
```
IR still on

vadimbz, 2021-12-07:
'''
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="NoIRCUTStartTime";'
|NoIRCUTStartTime|00:00:00
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="NoIRCUTEndTime";'
|NoIRCUTEndTime|00:01:00
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="NoIRLEDStartTime";'
|NoIRLEDStartTime|00:01:00
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_sys_param where c_param_name="NoIRLEDEndTime";'
|NoIRLEDEndTime|00:00:00
'''
No effect - IR still on.  Are these correct queries (straight quotes)?
'''
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value="00:00:00" where c_param_name="NoIRCUTStartTime";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value="00:01:00" where c_param_name="NoIRCUTEndTime";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value="00:01:00" where c_param_name="NoIRLEDStartTime";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value="00:00:00" where c_param_name="NoIRLEDEndTime";'
'''

roleoroleo, 2021-12-07:
Yes, the query is correct.

vadimbz, 2021-12-07:
Then it seems I'm out of options... other than wait for IRCUTMode to somehow kick in.

sanderbrauns, 2021-12-31:
Not sure what the settings mean. But with these settings the IR leds seem to turn off:

sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=2 where c_param_name="InfraredLamp";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=1 where c_param_name="LED1Mode";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=1 where c_param_name="LED2Mode";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=1 where c_param_name="IRCUTMode";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=0 where c_param_name="NoIRCUTStartTime";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=0 where c_param_name="NoIRCUTEndTime";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=0 where c_param_name="NoIRLEDStartTime";'
sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=0 where c_param_name="NoIRLEDEndTime";'

vadimbz, 2021-12-31:
> Not sure what the settings mean. But with these settings the IR leds seem to turn off:
Didn't work for me =(

rikki78, 2022-09-20:
> Not sure what the settings mean. But with these settings the IR leds seem to turn off:
>
> sqlite3 /mnt/mtd/db/ipcsys.db 'update t_sys_param set c_param_value=2 where c_param_name="InfraredLamp";' [...]

This worked for me

darkxst, 2022-09-26:
IRCUT will be an infrared filter that would normally be on during the day to cut out excess glare (IR from daylight/sunlight) from saturating the sensor. It would then turn off to allow night vision to work ;) I don't expect those settings to have any effect on the IR LED's

darkxst, 2022-10-01:
Can disable the visible status LED with the following blank file (reboot required) . 
```touch /dayun/mtd/cfg/disable_ledshow```

closed 2024-05-06 by github-actions[bot]