roleoroleo/sonoff-hack #74: Video Path

rcsmota, 2021-12-06:
Hi.

My motion detection videos are being recorded in /mnt/mmc/plan_record/ but the videos shown on the camera's motion event page are pointing to /mnt/mmc/alarm_record/.

So I can only access it via FTP but not on the camera page.

[image]

The files has this format. nrc20211206070627.mp4

Thanks

roleoroleo, 2021-12-07:
Probably they are in a different folder because are not generated by motion detection.

rcsmota, 2021-12-08:
Hello @roleoroleo.

Not being motion detection what could it be? From the videos it appears to be motion detection?
Isn't it because you have turned on motion detection by the camera browser (supported in firmware 0.1.1) instead of by the eWeLink software?

roleoroleo, 2021-12-08:
plan_record is used when you enable "Local record" with the app.

closed 2024-05-06 by github-actions[bot]