roleoroleo/sonoff-hack #76: MQTT issues - motion, birth will messages

eugeene, 2022-01-12:
I have the 'C' cam revision. Everything is fine. But I have two issues:

1. on boot the cam briefly sends MQTT '**online**' status and then sends and stays on '**offline**' Is it possible, the message settings are backwards?

2. the MQTT **motion_detection** is never sent - seems it is not working? Can i try some different setting?

Thank you!

roleoroleo, 2022-01-12:
Maybe there is a conflict between different mqtt connections.
Are you using mqtt advertise?
How many cams do you have?

eugeene, 2022-01-13:
I have only single camera. But many other tasmota devices (with completely different topics), so this should be irelevant.

roleoroleo, 2022-01-13:
Please, send me your mqtt.conf file.

eugeene, 2022-01-13:
Hello, It is basically the vanilla config, just changed the server ip and credentials.

`
###############################################################################
#                       Configuration file for mqtt-sonoff                    #
###############################################################################

# -----------------------------------------------------------------------------
# Basic MQTT settings
# -----------------------------------------------------------------------------

MQTT_IP=192.168.50.1
MQTT_PORT=1883

MQTT_CLIENT_ID=sonoff-cam

MQTT_USER=mqtt
MQTT_PASSWORD=********

MQTT_PREFIX=sonoffcam

# -----------------------------------------------------------------------------
# Set the topics where the messages will be published
# The final topic will be MQTT_PREFIX/TOPIC_MOTION
# -----------------------------------------------------------------------------

TOPIC_BIRTH_WILL=status
TOPIC_MOTION=motion_detection
TOPIC_MOTION_IMAGE=motion_detection_image

# -----------------------------------------------------------------------------
# Set the topics messages
# -----------------------------------------------------------------------------

BIRTH_MSG=online
WILL_MSG=offline
MOTION_START_MSG=motion_start
MOTION_STOP_MSG=motion_stop

# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN_BIRTH_WILL=1
MQTT_RETAIN_MOTION=1
MQTT_RETAIN_MOTION_IMAGE=1

`

roleoroleo, 2022-01-13:
Ok, the config is ok.
Try to kill the process mqtt-sonoff and restart it manually.
And post the output.

eugeene, 2022-01-14:
I changed literally nothing and the birth/will messages are working (for now) I guess there were retained messages on the server or something. But the motion sensing is still not working. Should id be enabled in the native eWlink app or is enabling it via web administration enough? 

Also with your hack, the camera appears online in the eWlink app, but I cant 'connect' to it - the stream is not visible (connecting to device, please waint), can't change any settings. I dont care about the native eWlink app, just sayin'

Thanks for your time

eugeene, 2022-01-14:
Ok, so I took the SD card out, enabled motion in the native app (had it enabled via the web interface only), put the SD card back and now the MQTT motion sensing seems to be working. No idea if this is supposed to be like that, but it's working!

So thank you for your time again ;)

closed 2022-01-14 by eugeene

roleoroleo, 2022-01-14:
I will check both the motion enable/°disable and app connection.
Thank you.