roleoroleo/sonoff-hack #77: Can I apply the firmware on a GK-200MP2B with a v28XXXXXX version???

oscarare, 2022-01-21:
Can I apply the firmware on a GK-200MP2B with a v28XXXXXX version???

roleoroleo, 2022-01-21:
I don't know, at the moment.
I have to test it.

DelusionalAI, 2022-01-23:
It worked for me.

riwich, 2022-02-26:
Not for me.. With Camera firmware v28xxxx and hack v0.09, everything worked fine. Since I updated to hack v0.1.1 I cannot longer connect to the camera with the eWeLink-App. Local RTSP-Stream and http OK. Of course, the Cloud-Functionality is enabled. Anyway, I had to reconnect the Cam to the eWeLink-App with a blank Micro SD-Card to use it in the eWeLink-App again... Unfortunately I cannot test it further at the moment, the camera is now running without the hack..

roleoroleo, 2022-03-01:
@riwich 
I'm trying to fix this problem.
Do you want to try a "patch"?
Overwrite system.sh
[system.sh.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/8162627/system.sh.tar.gz)

riwich, 2022-03-01:
Hello roleoroleo
Thank you very much for your efforts! Does the problem have to do with the
Sonoff Firware, or perhaps with new measures in the Sonoff Cloud? I would
very much like to test the patch, however I only have one "Sonoff
GK-200MP2B" available at the moment, and it has to run. I cannot experiment
with it... However, another "Sonoff GK-200MP2B" has been ordered and is on
its way. As soon as it arrives, I will be able to test it!

roleoroleo, 2022-03-01:
Ok, thank you.
I think the problem arises when I delete the admin user from the database.

riwich, 2022-03-01:
OK, soon I have my "Spare Cam", I will test it and report the result's!
Thanks again!

roleoroleo, 2022-03-02:
This one is better:
[system.sh.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/8168746/system.sh.tar.gz)

riwich, 2022-03-05:
My God, the Sonoff camera with its stupid sound setup is really bothering
me by now...
Unfortunately, the last version of the "system.sh" did not bring the
desired success. Even worse, with this version the camera seems to hang up
completely, there is no RTSP stream available on the local network, even
the websites are no longer accessible. By the way, the firmware version of
my camera is 28520210714.

roleoroleo, 2022-03-07:
Chek if the file is properly installed.
Windows/linux CR/LF...
Permissions...

riwich, 2022-03-07:
Hello roleoroleo!

Finally, it worked! The problem was my faulty Micro-SD card reader...
Meanwhile v0.1.1 with the updated "system.sh" works on both of my Sonoff
cameras. One camera runs with Sonoff firmware "27620201103", the other with
"28920211123".

Thank you for your work and patience!

Regards,
Thomas

kfulko, 2022-03-26:
Works with camera version '40620220309'

closed 2024-05-05 by github-actions[bot]