roleoroleo/sonoff-hack #78: MQTT Motion Detection without saving Video ?

stbuerger, 2022-02-04:
Is it actually possible to have motion detection and signaling it via MQTT - WITHOUT saving video files/clips when the detection is triggered ?
I've not yet found a way to achieve this. When I disable saving the video to SD card, I no longer get MQTT messages for motion detection events.
Is this not possible to do - or am I just doing it wrong ?

thx
-stefan

roleoroleo, 2022-02-12:
If you are not connected to the cloud, disable the switch "Recording without Cloud".
mp4record application will not start at all.

stbuerger, 2022-02-13:
> If you are not connected to the cloud, disable the switch "Recording without Cloud". mp4record application will not start at all.
THX a lot for this information. Just WHERE do I find this switch/setting ?

closed 2022-02-24 by stbuerger