roleoroleo/sonoff-hack #79: rtsp doesn't works

Dim3333, 2022-02-15:
Hi,

Firmware Version | 0.1.1
Base Version | V0525.1.276build20201103
Model | GK-200MP2C

Username and password are empty. I use this link:
[rtsp://hack:hack@MYIP/av_stream/ch0](url)

But connection refused with VLC. I don't know why ?

sccsltd, 2022-02-15:
Is your port set to 554 for rtsp? One of mine was set to something different set it back and it was fine.

Dim3333, 2022-02-15:
I used this link:
[rtsp://hack:hack@MYIP/av_stream/ch0](https://github.com/roleoroleo/sonoff-hack/issues/url)
and this one:
[rtsp://hack:hack@MYIP:554/av_stream/ch0](https://github.com/roleoroleo/sonoff-hack/issues/url)

Same issue.

sccsltd, 2022-02-15:
http://CAM-IP/index.html?page=configurations

on that page you can check the ports or change them if you want to 👍😊

Dim3333, 2022-02-15:
Yes, i do that.

neurocis, 2022-02-23:
I am having the same issue but do not even see a listener on 554:

`[root@GK]# netstat -a
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:1000            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:http            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN
tcp        0     64 172.27.1.129:ssh        172.27.1.141:57599      ESTABLISHED
`
I have tried changing ports for RTSP w/full resets to no avail. Stumped at this point. Thanks.

closed 2024-05-05 by github-actions[bot]