roleoroleo/sonoff-hack #80: MQTT (Mosquito Broker) under Home Assistant doesn't detect camera

Dim3333, 2022-02-16:
Hi,

I enabled MQTT service but under Home Assistant the camera doesn't appears in the MQTT Broker.
I well checked IP, login and password. I don't where is the issue.

Thanks.

roleoroleo, 2022-02-17:
Open a mqtt client (like MQTT-Explorer), connect to the broker and check if you can see the topics configured in the cam.
[image]

roleoroleo, 2022-02-17:
Try to restart the cam while the broker is running and check again.

Dim3333, 2022-02-17:
[image]

roleoroleo, 2022-02-17:
Try this binary.
I fixed a bug when the alarms table is empty.
[mqtt-sonoff.gz](https://github.com/roleoroleo/sonoff-hack/files/8092029/mqtt-sonoff.gz)

Dim3333, 2022-02-17:
It's seems better ....
[image]

But under Home Assistant the camera doesn't appears in the MQTT Broker.

Dim3333, 2022-02-18:
I use MQTT (Mosquito Broker) provided by HA
[image]

I already used your integration. Perfect +1
[image]

So I can disabled MQTT service in cam parameters ?

roleoroleo, 2022-02-18:
No, if you want to use my integration, mqtt is required.
But the cam does not appear in your mqtt broker because it doesn't use the discovery protocol.

roleoroleo, 2022-02-18:
Discovery requires a special protocol to be implemented on the cam.
I didn't write it.

darkxst, 2022-10-25:
Next release will have initial support for HA Discovery of the Camera Device info and the motion sensor and motion Image Sensor. 
It does not however implement all the features available in the Yi Hack integration.

bigsem89, 2023-09-09:
I'm having trouble getting my camera to work with MQTT. It seems that the MQTT topic isn't being created. I've already verified the correctness of the user, login credentials, IP address, and even checked the camera's network connectivity.

roleoroleo, 2023-09-20:
Please post your mqtt configuration (cam side).

bigsem89, 2023-09-21:
> Please post your mqtt configuration (cam side).

Thank you for response.
i have managed to fix it by changing event location to 0, but now i've faced another issue:

#156

closed 2024-05-05 by github-actions[bot]