roleoroleo/sonoff-hack #81: New firmware is not loading

attilabalazsy, 2022-02-16:
Hi,

I just bought a GK-200MP2-B camera exactly because it has this firmware hack. I've tried both firmwares (GK-200MP2B_0.1.1, GK-200MP2C_0.1.1), but non of them is loading, the camera always boots up to the original firmware. I've also tested with 2 different SD cards. Is there any option (instead of the serial connection) to get more information, why the firmware is not loading? I've checked the boot.sh and it starts by disabling the voice. My camera start up by these strange messages :). So I have a feeling the boot.sh is not executed ...

Thanks,
Attila

roleoroleo, 2022-02-17:
Which firmware version is your cam?
Check the app.

Dim3333, 2022-02-17:
[image]

attilabalazsy, 2022-02-18:
I see the following in the app:
Current version: 27620201103 (not sure what version is this :))
FW Name: GK-200MP2C

vertecx, 2022-02-19:
Try formatting your SD card in the camera by using the eWeLink app (Device Settings / More Settings / Format Card), then move it to your computer to copy the files, and then back again to the camera.

I could never get the hack to boot from an SD card formatted as FAT32 by Windows 10, but the same SD card formatted as FAT32 by the camera works flawlessly...

attilabalazsy, 2022-02-19:
It is working now. Most probably the issue was the same as @vertecx suggested, so thanks for your comment! Frankly I do not know what exactly helped as I put the same SD card in it (an older 1GB) which was already there earlier. I was trying different FWs and different cards, and it seems the last working option was never tried :) 
Anyway, the suggestion for the card formatting seems to me reasonable and most probably the other cards were incorrectly formatted as the FW was not event loaded from it. This old 1GB card was the first one, but there the FW version was wrong. So the FW version must be really checked in the eWeLink app as the box way saying something else :)

closed 2022-02-19 by attilabalazsy