roleoroleo/sonoff-hack #84: Customizing RTSP url for ONVIF for proxies

iointerrupt, 2022-02-27:
I have an RTSP proxy server (live555) running on a raspberry pi where I send all my RTSP camera strams.  The reason for this is that a lot of the cameras are on WiFi 2.4ghz and the OnVIF/RTSP clients are on 5ghz or hard-wired.  If we have mutiple clients directly accessing the RTSP/OnViF streams from the cameras, which are usually unicast, that creates additional redundent traffic on the 2.4ghz channel.  By having the live555 proxy on my PI, there is only one connection to the cameras as all the clients connect to the proxy for the stream and the stream.  Live555 automatically tears down the stream when the last client disconnects.  I noticed that the OnVIF daemon that loads, specifies the url for the video stream to use.  Can we customize the RTSP URL in the OnVIF config that gets used so it can point to a proxy?

roleoroleo, 2022-02-27:
If a script "/mnt/mmc/startup.sh" exists, the boot runs it at the end of the procecedure.
So you could create a startup.sh scripts that:
- kill onvif_srvd daemon
- create a custom /tmp/onvif_srvd.conf file with your custom entries
- restart onvif_srvd daemon

iointerrupt, 2022-02-27:
Perfect!  I'll just create a ```startup.sh``` that will kill the daemon, ```sed``` the temporary conf that and restart the daemon.

closed 2024-04-21 by iointerrupt