roleoroleo/sonoff-hack #89: very slow pan/tilt

iw2lsi, 2022-06-06:
Hi all...

  first of all,  thanks a lot for this great hack... it's really awesome!

  That said, am I the only one experiencing a very slow PT(Z) executions ? even when I use the embedded web-interface, it takes one minutes or more before the camera actually moves...

Looking the system status, it seems that I have less than 4k of free ram... should I enable the swap space ?

[image]

   thanks again,

        Giampaolo

iw2lsi, 2022-06-06:
BTW... I've tested also:

http://192.168.1.208:8080/cgi-bin/ptz.sh?dir=down

which does not work... and:

http://192.168.1.208/cgi-bin/ptz.sh?dir=down

that seems to works but again... it takes more tha 60s to execute a single movement...

roleoroleo, 2022-06-06:
Yes, try to enable swap space.

roleoroleo, 2022-06-06:
What happens if you try to move the cam with the app?

iw2lsi, 2022-06-06:
Hi...

you mean by using the web interface ? I'm not aware of a iOS/Android app...

 it's slow... and it appears to be slow even if I ssh into the busybox and
give a  command (eg: "ptz -a left") directly

I suspect that I have a network issue... but I cannot check it at the
moment...

        Giampaolo

ggggh, 2023-01-06:
Same issue here, but I'm sure it's not a network issue in my case.  At times it might take 20 seconds to execute a small movement.
Do you have an MP2C model?
I see your CPU usage is high, as mine; I think that's the problem.  Maybe pan movement is a particularly low priority task, and it takes a while to schedule that one.  I think our problem is the high cpu usage, and it shows especially on pan.
By the way, I would make sure you don't have an open browser on the PTZ tab of the web portal, as it consumes load of CPU for me with the constant snapshot updates.

roleoroleo, 2023-01-06:
You are right, the PTZ page is only a test page and consumes a lot of cpu because it calls a snapshot every second.
A snapshot means a lot of memory: 1920*1080*3 bytes.

closed 2024-05-05 by github-actions[bot]