roleoroleo/sonoff-hack #90: Test FTP connection

tukilo, 2022-06-24:
Hello there,

I am using a different port than usual to establish my FTP connection. Is there any way to check that the connection is working correctly through the web?

gewinh, 2022-06-24:
Don't think their is a way to test from web. but see "68" for commandline test.
Also think that ftppush.sh hard codes port 21 
 echo -e "USER ${FTP_USERNAME}\r\nPASS ${FTP_PASSWORD}\r\nmkd ${FTP_DIR}\r\nquit\r\n" | nc -w 5 ${FTP_HOST} 21 | grep "${FTP_DIR}"
                FTP_DIR="${FTP_DIR}/"
You could edit this file and try a different port

roleoroleo, 2022-06-28:
Try to run the command manually and check the output.

tukilo, 2022-06-28:
> Try to run the command manually and check the output.

yep, it actually worked by running

```
[root@GK]# echo -e "USER username\r\nPASS password\r\nmkd folder_name\r\nquit\r\n" | nc -w 5 local_ip_host 21 
220 FTP Server
331 Please specify the password.
230 Login successful.
257 "/folder_name" created
221 Goodbye.
```
then it is just a matter of understanding why it is not creating any records

roleoroleo, 2022-06-29:
I mean, try to run manually the ftppush.sh process.

tukilo, 2022-06-29:
I'm gonna reply myself.:
After changing some network config in my docker-compose file I made it to run successfuly and I could see the mock file on FTP side.

FTP transfer working so now I can only wonder why it is not recording anything

```
2022-06-29 [15-50-10] [INFO] checkFiles
257 "/01" created
257 "/01//01" created
2022-06-29 [15-50-13] [INFO] checkFiles: uploadToFtp SUCCEEDED - [/mnt/mmc/alarm_record/1970/01/01/ARC197100101020045.mp4].
/mnt/mmc/sonoff-hack/script/ftppush.sh: line 293: rmdir: not found
/mnt/mmc/sonoff-hack/script/ftppush.sh: line 293: rmdir: not found
/mnt/mmc/sonoff-hack/script/ftppush.sh: line 293: rmdir: not found
```

tukilo, 2022-06-30:
> Try to use the original app to check the configuration of the cam.

Indeed, looks like it needs some extra steps with the original app to finish the setup. 

Local record is enabled and motion sensitivy is high. Its creating a folder with today's date and thats it (/mnt/mmc/alarm_record/20220630)  no record file inside the folder

tukilo, 2022-07-13:
it's working now. I had to reformat the SD and start over again because something wasn't good. This time I started configuring the cam with the official app and afterwards I inserted the SD with the sonoff-hack. 
Thanks for your help :)

closed 2022-07-13 by tukilo