roleoroleo/sonoff-hack #93: PTZ not working

WildRat, 2022-08-04:
Could you, please, help me solve missing PTZ problem? Other functions are working. I try swapping ptz bin - camera self calibrates, but PTZ controls did not work. 

Reported at webpage info:
Firmware Version	0.1.2
Base Version	V5520.2053.0402build20220712
Model	
Device ID	a63001b3fb
In eWelink app i see firmware name as GK-200MP2D

I can make fylesystem dumps if needed.

roleoroleo, 2022-08-16:
Which control are you using to test PTZ?
onvif, web page, ...?

WildRat, 2022-08-16:
ONVIF. Web page show message PTZ not supported.

roleoroleo, 2022-08-18:
GK-200MP2D is not supported at the moment.
If you have a backup of the flash I could check if I can support it.

WildRat, 2022-08-18:
Please take a look when you have time. Here ([backup.zip](https://github.com/roleoroleo/sonoff-hack/files/9376182/backup.zip)) are the files obtained through the script (_Dump your backup firmware (SD card)_). And this is a new version of the firmware that they sent with corrected time zones: [Time zone issue Fix package.zip](https://github.com/roleoroleo/sonoff-hack/files/9376123/Time.zone.issue.Fix.package.zip).

roleoroleo, 2022-08-18:
Sorry but, after checking your backup, I found an error in the backup script at this page:
https://github.com/roleoroleo/sonoff-hack/wiki/Dump-your-backup-firmware-(SD-card))

Now, I updated it.
Please download it again and repeat the procedure.

WildRat, 2022-08-18:
I made a dump using the new version of the script: [backup_new.zip](https://github.com/roleoroleo/sonoff-hack/files/9376313/backup_new.zip).

roleoroleo, 2022-08-18:
Your cam is the same as in this post:
https://github.com/roleoroleo/sonoff-hack/issues/85

Sorry but I cannot support it, at the moment.

WildRat, 2022-08-18:
I understand, thanks. Apparently this is a new version of this model. If there is an opportunity to work on this version, I can help check the functionality.

roleoroleo, 2022-08-19:
To work on this version, we need to apply a lot of changes.
But a cam is needed.
And these changes must not break compatibility with the old version.
@puuu is working on it.

ConorIA, 2022-08-25:
This should be fixed in 0.1.3.
https://github.com/roleoroleo/sonoff-hack/issues/85#issuecomment-1227500921

WildRat, 2022-08-25:
But which version should I use?

roleoroleo, 2022-08-25:
Based on the backup you shared with me, use GK-200MP2-B.

roleoroleo, 2022-08-25:
@puuu @ConorIA 
Please reinstall 0.1.3 from scratch to avoid problem of future updates.

WildRat, 2022-08-25:
It's working without errors, but VERY slow. Video frame in 3-5 seconds. PTZ controls reacts once in 30-70 seconds.

roleoroleo, 2022-08-25:
Try ptz with the web service, not with the web page.
Because the web page calls a snapshot every second and the cam becomes slow.

WildRat, 2022-08-25:
I tried with ONVIF Device Manager - PTZ works great! But no live video. And strange video presets:
[image]
[image]

WildRat, 2022-08-25:
After couple reboots I managed to get it working with Synology Surveillance Station - video Ok, PTZ Ok. But strange video info:
[image]

bigsem89, 2022-08-30:
The same thing with  GK-200MP2B.  PTZ doesn't work. I've tried PTZ on http page and also via onvif manager.
it works with ewelink.

Firmware Version | 0.1.3

Base Version | V2524.1.245build20191030
Model | GK-200MP2B
Device ID | a63000ffc4

roleoroleo, 2022-09-01:
I will check it.

bigsem89, 2022-09-01:
> I will check it.

Thanks 🙏

roleoroleo, 2022-09-02:
Try to remove the file /mnt/mmc/sonoff-hack/bin/ptz
And reboot the cam.

bigsem89, 2022-09-04:
> Try to remove the file /mnt/mmc/sonoff-hack/bin/ptz And reboot the cam.

did so. 
same result. ptz still doesn't work.

roleoroleo, 2022-09-06:
Strange behavior.
Try to run the ws:
`http://IP_OF_YOUR_CAM/cgi-bin/ptz.sh?dir=right`

bigsem89, 2022-09-12:
> Strange behavior. Try to run the ws: `http://IP_OF_YOUR_CAM/cgi-bin/ptz.sh?dir=right`

{
"error":"false"
}

roleoroleo, 2022-09-12:
Does the cam moves?

roleoroleo, 2022-09-12:
And check the error running ipc_cmd with a ssh console.

bigsem89, 2022-09-13:
> Does the cam moves?

no

bigsem89, 2022-09-13:
> And check the error running ipc_cmd with a ssh console.

What is ipc_cmd? where can i find it?

roleoroleo, 2022-09-19:
Sorry, ipc_cmd is for another model. Try ptz command.
Login with ssh and run the following commands:
```
. /mnt/mmc/sonoff-hack/script/env.sh
ptz -a right
```

roleoroleo, 2022-12-14:
Please, check if you have a library `libhardware.so` or `libptz.so` in the folder `/mnt/mtd/ipc/app/lib`

officialjonot, 2022-12-14:
`[root@sonoff-c00d]# cd /mnt/mtd/ipc/app/lib
[root@sonoff-c00d]# ls
libNetLib.so            libdbmanager.so         libserviceinterface.so
libappmanager.so        libhardware.so          libservicemanager.so
libavcodec.so.55        libioport.so            libspeaker.so
libavformat.so.55       libled.so               libspi.so
libavsharemem.so        liblogmanager.so        libssl.so
libavutil.so.52         libmainic.so            libssl.so.1.0.0
libcfg.so               libportmap.so           libusermanager.so
libcommunication.so     libptz.so               libwdt.so
libcrypt-0.9.33.2.so    librtc.so               libwifi.so
libcrypt.so.0           librtmp.so              libz.so.1
libcrypto.so            libsd.so
libcrypto.so.1.0.0      libsensor.so`

officialjonot, 2022-12-14:
```
[root@sonoff-c00d]# ptz_h -a right
ptz_h: can't resolve symbol 'hw_ptz_sendptz'
```

roleoroleo, 2022-12-14:
Ok, copy the file:
`cp /mnt/mmc/sonoff-hack/bin/ptz_p /mnt/mmc/sonoff-hack/bin/ptz`

officialjonot, 2022-12-14:
ok fantastic - its working on the web PTZ panel now (http://<IP-OF-CAMERA>/?page=ptz).  Iwill invesitgate to see if it works on ONVIF device manager. Thank you Roleo.

EpicLPer, 2023-01-26:
> Ok, copy the file: `cp /mnt/mmc/sonoff-hack/bin/ptz_p /mnt/mmc/sonoff-hack/bin/ptz`

Just wanted to say that this also worked for mine :)

closed 2024-05-05 by github-actions[bot]