roleoroleo/sonoff-hack #102: Home Assistant mqtt changes

darkxst, 2022-09-18:
There were some breaking changes that alter mqtt configurations in Home Assistant 2022.06. Support for old format will be completely removed soon. Examples on the wiki can be updated to the following for new format (and a minor fix for payload values).

```
mqtt:
  binary_sensor:
    - state_topic: "home-assistant/kitchen/motion_detection"
      name: "Motion detection - kitchen"
      payload_on: "motion_start"
      payload_off: "motion_stop"
      qos: 1
```

```
mqtt:
  camera:
    - topic: "home-assistant/kitchen/motion_detection_image"
      name: "Motion detection image - kitchen"
```

roleoroleo, 2022-09-20:
Wiki updated.

darkxst, 2022-09-20:
Thanks, will close this now.

closed 2022-09-20 by darkxst