roleoroleo/sonoff-hack #104: pulling the motion_detection_image off MQTT and into a notification (home assistant)

keiranharris, 2022-09-22:
Hi, i appreciate this is not a bug - but ive ran dry with my googling, and figure as you wrote the code there is a fair chance you will know the direction to point me in. 

im running your hack and loving it. Today i ventured into MQTT world (first time ive needed it, although im a long term home assistant user). I went here, to get motion detect images from these sonoff cams, sent to my telegram. 

My early automation is below - and it works just fine - as in when i walk past the cam, my telegram alerts me, with a simple message:  
[image]

but how do i get the motion_detection_image snapshot (that i can see is being published) off the MQTT message bus, and into my notifiy automation action above? 
[image]
 
thanks in advance! 
great work BTW! 
Keiran.

roleoroleo, 2022-09-22:
I don't know exactly how to help you.
But if you are a python user, you could read the code of the mqtt cam in homeassistant/core
Maybe this component uses the same approach.

keiranharris, 2022-09-22:
Ok thanks i can try on the weekend. BTW - i have seen other people, on non sonoff cams, solve this by 1/ invoking the home assistant cam snapshot function 2/ writing that image to the local filesystem somewhere 3/ then picking up that latest image to embed in a notification message..... but give you ALREADY put an image on the bus, this seems inefficient to me. 

Out of curiosity, outside of home assistant (which sounds like you might not use?), do you do this? Use that image you put on the mqtt bus, and do _something_ with it?

darkxst, 2022-09-25:
You should be able to get the image from the MQTT message from the  ```trigger.payload``` variable in your automation. However telegram can only read images from a file or URL so you probably going to have to save it anyway.

If you are using the MQTT camera integration, then the following will probably work to get the image also:
https://www.home-assistant.io/integrations/generic/#sharing-a-camera-feed-from-one-home-assistant-instance-to-another

josecarre, 2022-09-27:
@keiranharris here you have my automation just in case it helps you. When motion is detected it send me a telegram msg with the image 😃  :
```
alias: Send Sonoff camera snapshot to Telegram when motion is detected
description: ""
trigger:
  - platform: state
    entity_id:
      - binary_sensor.yi_hack_s_93abe8_motion_detection
    from: "off"
    to: "on"
condition: []
action:
  - service: camera.snapshot
    data:
      filename: /config/www/tmp/{{filename}}.jpg
    target:
      entity_id: camera.yi_hack_s_93abe8_cam
  - service: notify.pixel
    data:
      message: Movimiento detectado
      data:
        message: Movimiento detectado
        photo:
          - file: /config/www/tmp/{{filename}}.jpg
            caption: 🚨 Movimiento detectado en salón {{filename}}
variables:
  filename: snapshot-{{now().strftime("%d%m-%H%M%S")}}
mode: single
```

closed 2024-05-05 by github-actions[bot]