roleoroleo/sonoff-hack #107: timezone not updated

josecarre, 2022-09-26:
Hi, i've installed one new camera:
```
Firmware Version | 0.1.3
-- | --
Base Version | V5520.2053.0383build20220226
Model | GK-200MP2-B
```
Not sure why... but even I set this timezone in conf page: Europe/Amsterdam

I still see the time wrong...

Then I entered through SSh and i found the issue:
```
[root@GK]# cat /etc/TZ
EST-8:00:00
```

Not sure why is not following the ocnfig from the web :(

darkxst, 2022-09-27:
I dont think that file is used, the timezone is stored in the sqlite database. On my camera time is correct, even though TZ file is wrong.

```
[root@GK]# cat /etc/TZ
EST-10:00:00

[root@GK]# date -R
Tue, 27 Sep 2022 19:01:28 +1000

[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_sys_param where c_param_name='ZoneTimeName';"
|ZoneTimeName|71
```

josecarre, 2022-09-27:
Hmmmm, not sure why this is happening. In the UI I have Europe/Amsterdam... but seems not to be followed

Could it be related to the base version? This camera I've bought is pretty new (V5520.2053.0383build20220226), but I have other one that work like a charm since a lot of time with V2524.1.245build20191030. Both cameras are same model GK-200MP2-B

Maybe downgrade fixes it? or not possible? I'm seeing other issues, I just installed the new cam yesterday but this DATE thing is the first one i clearly found as a wrong thing.

darkxst, 2022-09-27:
maybe your older one is the GK-200MP2B, and you new one the GK-200MP2-B? They are 2 different models. There are some differences in those firmwares, including that ```/etc``` is read-only on the newer firmware.

You could try upgrading to ```V5520.2053.0402build20220712``` that is the firmware which I am running and timezones work.

josecarre, 2022-09-27:
Oh! you;re right, I didnt see it. The old cam is GK-200MP2B, and new one is GK-200MP2-B... ouch! Anyway for this new one I applied the right sonoff-hack file GK-200MP2-B_0.1.3.tgz

You mean upgrading the sonoff firmware **to the last one** with ewelink app? Im scared when upgrading firmwares hehehe, is the rest also working? I've seen more issues: https://github.com/roleoroleo/sonoff-hack/issues/108

darkxst, 2022-09-27:
> You mean upgrading the sonoff firmware **to the last one** with ewelink app? Im scared when upgrading firmwares hehehe, is the rest also working? I've seen more issues: #108

Yes, but I am being offered a newer build again in ewelink app now, which I havent tested either.

MQTT works with my PR. There were reports of rtsp streams crashing in #85 . I found the yi-hack HA integration was causing my camera (S-CAM model) to crash, but dont see those crashes when using the ONVIF integration.

josecarre, 2022-09-27:
Yeah, ewelink app offers me `41120220811`, seems much newer, from august. I will try to solve the issues for now with the firmware I have `38320220226`, at least till @roleoroleo confirms that we can upgrade. But usually im not a friend of upgrading official firmware... unless something is not working

I've tested your PR build, and the TZ remains wrong :(

darkxst, 2022-09-27:
> I've tested your PR build, and the TZ remains wrong :(

That should fix MQTT though, there is nothing in that build likely to change TZ behaviour vs 0.1.3 release

darkxst, 2022-09-27:
timezone appears to be stored here
```/dayun/mtd/db/conf/config_timezone.ini```

Changing that value will override the timezone, however not sure where its got my settings from originally maybe the eWeLink app?

darkxst, 2022-09-27:
> Yeah, ewelink app offers me `41120220811`,

but that is only slightly newer than the July build that has been tested

josecarre, 2022-09-27:
Hi, sorry for the delay, it took me some time to test this and the other thread. I have:
- removed camera from ewelink app and factory reset camera
- add again in app, and set timezone un app
- upgraded to `41120220811`
- then applied your sonoff-hack built based on 0.1.3

And the timezone thing seems fixed :)

Not sure if setting the TZ first in the app was needed to be honest, I didnt do anything on the app but connecting camera to wifi, so maybe some additional stuff like this was needed.

@roleoroleo I confirm 0.1.3 built from @darkxst works in `41120220811`, just in case you want to update supported versions in readme.md

closed 2022-09-28 by josecarre

roleoroleo, 2022-09-30:
> timezone appears to be stored here `/dayun/mtd/db/conf/config_timezone.ini`
> 
> Changing that value will override the timezone, however not sure where its got my settings from originally maybe the eWeLink app?
> 
> It doesnt seem to follow the timezone settings in the Sqlite database at all

What happens when you change timezone inside the app?
Does the row in the db change?
Does config_timezone.ini change?

Probably I need to update https://github.com/roleoroleo/sonoff-hack/blob/master/src/www/httpd/cgi-bin/set_configs.sh

darkxst, 2022-10-01:
> What happens when you change timezone inside the app?

It does not seem to have any affect , even with 'sync time' under camera settings.

So did a factory Reset and Re-paired....
> Does the row in the db change?

No it remains set at the default of 61, which is Hong Kong time
> Does config_timezone.ini change?

This appears to updated during the pairing process, changing the timezone after pairing has no effect on this file or the camera time zone.

josecarre, 2022-10-02:
From my first post:
> [root@GK]# cat /etc/TZ
> EST-8:00:00

I see that now this file changed, so it seems to have effect, now:
```
[root@GK]# cat /etc/TZ
EST-2:00:00
```

darkxst, 2022-10-13:
I just got my hands on a sample of GK-200MP2-B camera. as noticed by @josecarre there are major bugs in timezone handling on this with factory firmware version! Update to august firmware + re-pairing seems required to fix it. 

at which point ```/etc/TZ``` is correct, and happily accepts changes. Should be able to just change that via web UI and changes will take.

darkxst, 2022-10-14:
Actually ```/etc/TZ``` wont be saved its in the ramdisk. Will be lost on reboot.
Which leaves us with only ```/dayun/mtd/db/conf/config_timezone.ini```
The timezone setting in the db are not used on this firmware.

roleoroleo, 2022-10-14:
So the path is:
App -> config_timezone.ini -> /etc/TZ
Right?

darkxst, 2022-10-15:
The App sets config_timezone.ini when pairing the device. 
/etc/TZ is not directly linked to this config file, however is set by the devctl process running on the Camera.
Manually changing config_timezone.ini and restarting devctrl updates /etc/TZ

Changing the timezone of the phone that is running the App, and then selecting "More Settings -> Sync Time" updates timezone in config_timezone.ini and /etc/TZ (without reboot or restart).

Changing the timezone setting in the App seems to do nothing.

roleoroleo, 2022-10-16:
I could change set_configs.sh to change config_timezone.ini

darkxst, 2022-10-16:
yes, that seems to be the only option on newer firmware, but worth checking behaviour on previous models also. I dont have any of those to test here!

gewinh, 2023-05-16:
Not sure where the above issue stands at the moment. But I have finally found some time to look at a related problem, re changing the time when daylight savings time changes. I built some GK-200MP2C cameras with 0.1.5 both before daylight saving changed and some after. The ones before the change have the incorrect time ie +1hr, the ones after the change have the correct time. So the initial setup gets the right time from the timezone, however, the timezone offset isn't updated when daylight savings changes. Going into the webui and resetting timezone doesn't fix the problem. The only way I have found to get the right time on the cameras when daylight savings changes  is to edit /mnt/mtd/cfg/config_timezone.ini . I am not sure at what stage the value in this file is generated, but it would be nice if the webui value was updated when the timezone value changes.

roleoroleo, 2023-05-16:
What do you mean exactly?
The cam doesn't update when the time passes from no DST to DST?
Or when you are in DST the time is incorrect?

What I mean is: if you reboot the cam when the time is in DST, is the time correct?

darkxst, 2023-05-17:
> So the initial setup gets the right time from the timezone,

For initial setup, eWeLink app syncs the timezone with your phone as a one off operation. I dont think there is any support for daylight savings rules to automatically change for DST

gewinh, 2023-05-17:
Sorry for the confusion, yes the cam doesn't update when the time passes from no DST to DST. I haven't figured out the workflow for resetting TZ but I notice that webui timezone data isn't stored in /sonoff-hack/etc/system.conf, the timezone parameter in there is blank. So when env.sh  runs ,TZ isn't set.  If I manually change TZ the time is changed until reboot. I will, if I can get some time(no pun intended), experiment with manually  setting timezone in system.conf and manually running env.sh to see if that would fix it on reboot, the outstanding issue then would be why Timezone isn't updated in system.conf from the webui value.

gewinh, 2023-05-17:
OK a little bit more info.  If I have /mnt/mtd/cfg/config_timezone.ini set to 39600 ie 11hrs then edit mnt/mmc/sonoff-hack/etc/system.conf and include a timezone value of Australia/Sydney ie 10hrs into that parameter, then reboot the time offset is still 11hrs  and the value of TZ in /etc/TZ is still 11hrs. I am guessing that as darkxst suggests, the initial setup sets the value in config_timezone.ini which sets /etc/TZ on boot. I was hoping that a value in system.conf and running env.sh after the boot might update TZ. If I edit the /etc/TZ file and change the timezone that value changes the clock until reboot. I would guess that if the webui put the value in system.conf then on boot env.sh ran and update /etc/TZ then that would set the correct time on the clock regardless of the hardcoded value in config_timezone.ini. All this assumes there is some mechanism of reading TIMEZONE offsets for a  given setting on boot if this isn't happening on boot then DST, nonDST settings will not be taken into account. I think I have reach the limit of my understanding on home all this fits together.

roleoroleo, 2023-05-17:
TIMEZONE field in system.conf is a typo related to an old version.
I will remove it in the next release.
The timezone is saved directly into the database in the same way the app does.
So, the hack doesn't change the behavior of the original fw.

I agree with darkxst, DST is not handled by Sonoff.

darkxst, 2023-05-17:
> I haven't figured out the workflow for resetting TZ

If you do this in the eWeLink app -> "More Settings -> Sync Time" it will update the camera timezone (via config_timezone.ini) to match your phones timezone. 

Some newer firmwares do not use the value stored in the database anymore. However, not sure if GK-200MP2C  is affected by this.

gewinh, 2023-05-17:
Thanks Roleo,Darkxst, I agree, this one needs to go into the too hard basket. I have looked at 3 different firmware and found there is a different way of handling TZ in each case.  config_timezone.ini isn't present in 2 out of 3 of these firmwares (not sure if this is because I didn't Sync Time with the app) .  I will check if I can get this file into the system for other firmwares if I use the app to sync time. I have a nightly crontab job that runs on a server and backups mp4's from the camera. If I can get config_timezone.ini working on the other firmwares I will add some bash to my backup job to compare the current timezone with the value in config_timezone.ini, if it is different I will update the file. A low priority job for me now  though.
If anyone is interested here is the initial bash script that I have been playing with.
#!/bin/sh
hr_offset=$(date +%:::z|sed 's/+0//')
echo $hr_offset
sec_offset=$(($hr_offset * 3600))
echo $sec_offset
#
#Compare values and update config_timezone.ini if necessary
#
cam_sec_offset=$(sed -n '2p' /filecopied via scp to a temp working area/config_timezone.ini|sed 's/offset_second=//'|sed 's/"//'|sed 's/"//')
echo $cam_sec_offset
if [ "$cam_sec_offset" -ne "$sec_offset" ]; then
  echo "Two valies not equal updating file"
  sed -i -e "s/$cam_sec_offset/$sec_offset/g" /filecopied via scp to a temp working area/config_timezone.ini;
#  copy new config_timezone.ini back to camera
fi

roleoroleo, 2023-08-19:
Do you want to try this new script?
It should fix the timezone issue.
[set_configs.sh.gz](https://github.com/roleoroleo/sonoff-hack/files/12386390/set_configs.sh.gz)

jezzaaa, 2024-07-30:
It's almost a year later, and there has been no feedback.

I can confirm that this replacement script fixed the problem for me.

Thanks so much @roleoroleo. I've sent you a donation, for your continuing work to help others make the most of their cameras.

roleoroleo, 2024-07-30:
Thank you for your feedback.
Change commited: https://github.com/roleoroleo/sonoff-hack/commit/d676f8465ea0ba04eb6f8e1170f97922e6287992
And thank you for your donation!

commit d676f846 (2024-07-30): Fix #107