roleoroleo/sonoff-hack #108: GK-200MP2-B and V5520 with MQTT and other issues

josecarre, 2022-09-27:
Hi,

I have read other issues that could be related:
https://github.com/roleoroleo/sonoff-hack/issues/69
https://github.com/roleoroleo/sonoff-hack/issues/80
And this one mine, that maybe is also affected with something bigged than just TZ thing:
https://github.com/roleoroleo/sonoff-hack/issues/107

I see a PR that is coming to fix any or all of this? https://github.com/roleoroleo/sonoff-hack/pull/103

The thing is that I always had a GK-200MP2-B at home, with V2524.1.245build20191030
And now I have a new one, same model, but base version is V5520.2053.0383build20220226
On this new one I installed directly last version 0.1.3, but the other came from other versions from a year ago or so... updating...

Not sure if this is related to SW... or HW.... or Base. version... but... For thr MQTT config it seems that is doing nothing, or im not applying the right configs... Because in mqtt-explorer i dont see what i expect
Some basic questions first:
WIth two cameras...
1. should i set the same "Topics Prefix" in MQTT config page? Or each camera should have its own topic prefix?
2.  The only thing i have with different value is MQTT "Client ID". I understand that this is the way each device will identify itself when sending msgs? or not?
One I understand at least if the topic should be the same or not (for yi-hack HA integration to work fine) then i will advance a bit in tbshoot this :)

Not sure if downgrading base version SW is an option... but the old cam works very well :(

darkxst, 2022-09-27:
Yes my PR should fix MQTT on your new camera. You can try test that with the Github action builds here (or just grab the mqtt-sonoff binary from that build and put on your camera)
https://github.com/roleoroleo/sonoff-hack/actions/runs/3134840557

Re configs
1. MQTT Topic Prefix needs to be unique for each camera, Home Assistant uses this one to identify the camera.
2. Client ID needs to be unique also but I believe this is only used by MQTT broker and not in Home Assistant.
3. hostname handling should be the same for both cameras using ```/mnt/mmc/sonoff-hack/etc/hostname``` or web config. they should be different but depending on your router may not be resolvable anyway.

As a side note I am looking at adding MQTT discovery messages for Home Assistant and will likely use the "Client ID" as the unique Id for these messages.
https://www.home-assistant.io/docs/mqtt/discovery/

darkxst, 2022-09-27:
Try run the following in ssh and see if it logs motion events. I wonder if your timezone issue causing other issues also!
```
[root@GK]# . /mnt/mmc/sonoff-hack/script/env.sh 
[root@GK]# mqtt-sonoff
```

darkxst, 2022-09-27:
also try run this one
``` sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_rec_dbfile;' ```

josecarre, 2022-09-27:
I tried yesterday some stuff from other thread... but seems like sqlite was not there to be used:
```
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_sys_param where c_param_name='ZoneTimeName';"
-sh: sqlite3: not found
```

josecarre, 2022-09-27:
Hi @darkxst , all that stuff should be ok now, cannot go back in time and troubleshoot because I reset camera and redo all installation (details here https://github.com/roleoroleo/sonoff-hack/issues/107#issuecomment-1259572531)

The thing now is that I see less msgs for the new cam... and the. motion detection doesnt work :(

Old camera is sonoffcam, and new one is sonoffcamgarage:
[image]
There must be a reason not to see motion detection stuff, right?
Thats one thing... other minor thing is to see sonofcam offline while its perfectly online... not sure if its related to QoS of msgs... or any retain thing, but cameras are up and working.

First camera (old one), has this config:
```
"MQTT_IP":"192.168.1.101",
"MQTT_PORT":"1883",
"MQTT_CLIENT_ID":"sonoff-cam",
"MQTT_USER":"*****",
"MQTT_PASSWORD":"******",
"MQTT_PREFIX":"sonoffcam",
"TOPIC_BIRTH_WILL":"status",
"TOPIC_MOTION":"motion_detection",
"TOPIC_MOTION_IMAGE":"motion_detection_image",
"BIRTH_MSG":"online",
"WILL_MSG":"offline",
"MOTION_START_MSG":"motion_start",
"MOTION_STOP_MSG":"motion_stop",
"MQTT_KEEPALIVE":"120",
"MQTT_QOS":"1",
"MQTT_RETAIN_BIRTH_WILL":"1",
"MQTT_RETAIN_MOTION":"0",
"MQTT_RETAIN_MOTION_IMAGE":"0",
"MOTION_IMAGE_DELAY":"0.5",
"NULL":"NULL"
```

And the new cam:
```
"MQTT_IP":"192.168.1.101",
"MQTT_PORT":"1883",
"MQTT_CLIENT_ID":"sonoff-cam-garage",
"MQTT_USER":"*******",
"MQTT_PASSWORD":"******",
"MQTT_PREFIX":"sonoffcamgarage",
"TOPIC_BIRTH_WILL":"status",
"TOPIC_MOTION":"motion_detection",
"TOPIC_MOTION_IMAGE":"motion_detection_image",
"MOTION_IMAGE_DELAY":"0.5",
"BIRTH_MSG":"online",
"WILL_MSG":"offline",
"MOTION_START_MSG":"motion_start",
"MOTION_STOP_MSG":"motion_stop",
"MQTT_KEEPALIVE":"120",
"MQTT_QOS":"1",
"MQTT_RETAIN_BIRTH_WILL":"1",
"MQTT_RETAIN_MOTION":"0",
"MQTT_RETAIN_MOTION_IMAGE":"0",
"MQTT_IPCSYS_DB":"0",
"NULL":"NULL"
```

darkxst, 2022-09-28:
> There must be a reason not to see motion detection stuff, right?

1. Do you see motion events in the events page on the webui? If that is not working MQTT wont work
2. Next check that you are seeing motion events in the sqlite db per above

> Thats one thing... other minor thing is to see sonofcam offline while its perfectly online... not sure if its related to QoS of msgs... or any retain thing, but cameras are up and working.

The cam will show offline until it receives a motion event (unless you have setup an availability_topic on the sensors in HA, pointing at the status topic). Qos and retain settings should be perfectly fine at the defaults.

josecarre, 2022-09-28:
> 1. Do you see motion events in the events page on the webui? If that is not working MQTT wont work

Where is that on the sonoff-hack GUI? Im blind 🤣

darkxst, 2022-09-28:
> Where is that on the sonoff-hack GUI? Im blind

Its the icon with two arrows pointing opposite directions, in the middle of the menu bar.
http://<IP>/?page=eventsdir

Then scroll down to the very bottom of that page

josecarre, 2022-09-28:
Aaaaa, you mean the videos recorded, ok ok, yes, I see them, one folder per hour, with several videos when motion detected yeah :)
It seems to work internally then, right? Not sure why is not sending right mqtt msgs :( , im using your custom firmware right now... should i go back to official one? Pressing the upgrade to 0.1.3 button should do it i guess. Or better to stay o this one?

darkxst, 2022-09-28:
now check the db

do you have this file?
```/mnt/mmc/AVRecordFile.db```

if so run the below, which should show a list of events, that match up whats on the events page
```
[root@GK]# . /mnt/mmc/sonoff-hack/script/env.sh
[root@GK]# sqlite3 /mnt/mmc/AVRecordFile.db 'select * from T_RecordFile;'
```
You probably need to stay on my patched firmware, unless the following command show a list of events (but guessing it wont on new firmware unless they changed it in that last build)
```
[root@GK]#sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_alarm_log;'

darkxst, 2022-09-28:
Also just double check that config is correct, as below:
```
[root@GK]# tail -n 1 /mnt/mmc/sonoff-hack/etc/mqtt-sonoff.conf 
MQTT_IPCSYS_DB=0
```

josecarre, 2022-09-28:
Thanks, recordfile is showing the videos of eventsdir... but alarm log seems empty... is the alarm part the one not triggering right mqtt messaging?

```
[root@GK]# ls /mnt/mmc/AVRecordFile.db
/mnt/mmc/AVRecordFile.db
[root@GK]# . /mnt/mmc/sonoff-hack/script/env.sh
[root@GK]# sqlite3 /mnt/mmc/AVRecordFile.db 'select * from T_RecordFile;'
20220927|2022-09-27 15:05:25|2022-09-27 15:06:03|2|ARC20220927150523.mp4|/mnt/mmc/alarm_record/20220927/15/ARC20220927150523.mp4|0||2022-09-27 15:05:24|
20220927|2022-09-27 15:11:25|2022-09-27 15:11:59|2|ARC20220927151124.mp4|/mnt/mmc/alarm_record/20220927/15/ARC20220927151124.mp4|0||2022-09-27 15:11:25|
20220927|2022-09-27 15:20:14|2022-09-27 15:20:49|2|ARC20220927152013.mp4|/mnt/mmc/alarm_record/20220927/15/ARC20220927152013.mp4|0||2022-09-27 15:20:15|
20220927|2022-09-27 15:28:21|2022-09-27 15:28:56|2|ARC20220927152819.mp4|/mnt/mmc/alarm_record/20220927/15/ARC20220927152819.mp4|0||2022-09-27 15:28:20|
20220927|2022-09-27 15:45:27|2022-09-27 15:45:58|2|ARC20220927154526.mp4|/mnt/mmc/alarm_record/20220927/15/ARC20220927154526.mp4|1||2022-09-27 15:45:59|
20220927|2022-09-27 15:45:57|2022-09-27 15:46:35|2|ARC20220927154556.mp4|/mnt/mmc/alarm_record/20220927/15/ARC20220927154556.mp4|1||2022-09-27 15:45:59|
20220927|2022-09-27 15:47:28|2022-09-27 15:48:02|2|ARC20220927154730.mp4|/mnt/mmc/alarm_record/20220927/15/ARC20220927154730.mp4|1||2022-09-27 15:47:31|
20220927|2022-09-27 15:59:11|2022-09-27 15:59:46|2|ARC20220927155912.mp4|/mnt/mmc/alarm_record/20220927/15/ARC20220927155912.mp4|1||2022-09-27 15:59:13|
20220927|2022-09-27 16:01:33|2022-09-27 16:02:06|2|ARC20220927160132.mp4|/mnt/mmc/alarm_record/20220927/16/ARC20220927160132.mp4|1||2022-09-27 16:01:35|
20220927|2022-09-27 16:19:38|2022-09-27 16:20:14|2|ARC20220927161938.mp4|/mnt/mmc/alarm_record/20220927/16/ARC20220927161938.mp4|0||2022-09-27 16:19:39|
20220927|2022-09-27 16:22:42|2022-09-27 16:23:17|2|ARC20220927162243.mp4|/mnt/mmc/alarm_record/20220927/16/ARC20220927162243.mp4|1||2022-09-27 16:23:17|
20220927|2022-09-27 16:23:16|2022-09-27 16:23:49|2|ARC20220927162314.mp4|/mnt/mmc/alarm_record/20220927/16/ARC20220927162314.mp4|1||2022-09-27 16:23:17|
20220927|2022-09-27 16:47:25|2022-09-27 16:47:59|2|ARC20220927164724.mp4|/mnt/mmc/alarm_record/20220927/16/ARC20220927164724.mp4|1||2022-09-27 16:47:27|
20220927|2022-09-27 17:02:08|2022-09-27 17:02:41|2|ARC20220927170208.mp4|/mnt/mmc/alarm_record/20220927/17/ARC20220927170208.mp4|1||2022-09-27 17:02:10|
20220927|2022-09-27 17:07:22|2022-09-27 17:07:57|2|ARC20220927170723.mp4|/mnt/mmc/alarm_record/20220927/17/ARC20220927170723.mp4|1||2022-09-27 17:07:24|
20220927|2022-09-27 17:09:18|2022-09-27 17:09:51|2|ARC20220927170919.mp4|/mnt/mmc/alarm_record/20220927/17/ARC20220927170919.mp4|1||2022-09-27 17:09:20|
20220927|2022-09-27 17:53:10|2022-09-27 17:53:43|2|ARC20220927175311.mp4|/mnt/mmc/alarm_record/20220927/17/ARC20220927175311.mp4|1||2022-09-27 17:53:12|
20220927|2022-09-27 17:56:20|2022-09-27 17:56:53|2|ARC20220927175620.mp4|/mnt/mmc/alarm_record/20220927/17/ARC20220927175620.mp4|1||2022-09-27 17:56:22|
20220927|2022-09-27 18:00:57|2022-09-27 18:01:31|2|ARC20220927180058.mp4|/mnt/mmc/alarm_record/20220927/18/ARC20220927180058.mp4|1||2022-09-27 18:00:59|
20220927|2022-09-27 18:05:02|2022-09-27 18:05:37|2|ARC20220927180504.mp4|/mnt/mmc/alarm_record/20220927/18/ARC20220927180504.mp4|1||2022-09-27 18:05:05|
20220927|2022-09-27 18:08:14|2022-09-27 18:08:47|2|ARC20220927180813.mp4|/mnt/mmc/alarm_record/20220927/18/ARC20220927180813.mp4|1||2022-09-27 18:08:16|
20220927|2022-09-27 18:56:16|2022-09-27 18:56:49|2|ARC20220927185616.mp4|/mnt/mmc/alarm_record/20220927/18/ARC20220927185616.mp4|1||2022-09-27 18:56:17|
20220927|2022-09-27 19:31:18|2022-09-27 19:31:53|2|ARC20220927193120.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927193120.mp4|1||2022-09-27 19:31:21|
20220927|2022-09-27 19:31:53|2022-09-27 19:32:27|2|ARC20220927193154.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927193154.mp4|1||2022-09-27 19:32:27|
20220927|2022-09-27 19:32:25|2022-09-27 19:32:59|2|ARC20220927193226.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927193226.mp4|1||2022-09-27 19:32:27|
20220927|2022-09-27 19:34:55|2022-09-27 19:35:29|2|ARC20220927193457.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927193457.mp4|1||2022-09-27 19:34:58|
20220927|2022-09-27 19:35:39|2022-09-27 19:36:13|2|ARC20220927193540.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927193540.mp4|1||2022-09-27 19:35:42|
20220927|2022-09-27 19:39:11|2022-09-27 19:39:45|2|ARC20220927193913.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927193913.mp4|1||2022-09-27 19:39:14|
20220927|2022-09-27 19:40:38|2022-09-27 19:41:11|2|ARC20220927194039.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927194039.mp4|1||2022-09-27 19:40:41|
20220927|2022-09-27 19:51:14|2022-09-27 19:51:49|2|ARC20220927195116.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927195116.mp4|1||2022-09-27 19:51:17|
20220927|2022-09-27 19:54:57|2022-09-27 19:55:31|2|ARC20220927195459.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927195459.mp4|1||2022-09-27 19:55:00|
20220927|2022-09-27 19:55:42|2022-09-27 19:56:15|2|ARC20220927195543.mp4|/mnt/mmc/alarm_record/20220927/19/ARC20220927195543.mp4|1||2022-09-27 19:55:44|
20220927|2022-09-27 20:56:38|2022-09-27 20:57:13|2|ARC20220927205640.mp4|/mnt/mmc/alarm_record/20220927/20/ARC20220927205640.mp4|1||2022-09-27 20:56:41|
[... cut ...]
20220928|2022-09-28 09:10:57|2022-09-28 09:11:32|2|ARC20220928091057.mp4|/mnt/mmc/alarm_record/20220928/09/ARC20220928091057.mp4|1||2022-09-28 09:10:58|
20220928|2022-09-28 10:27:40|2022-09-28 10:28:14|2|ARC20220928102741.mp4|/mnt/mmc/alarm_record/20220928/10/ARC20220928102741.mp4|1||2022-09-28 10:27:42|
[root@GK]#
[root@GK]#
[root@GK]# sqlite3 /mnt/mtd/db/ipcsys.db 'select * from t_alarm_log;'
[root@GK]#
[root@GK]#
[root@GK]# tail -n 1 /mnt/mmc/sonoff-hack/etc/mqtt-sonoff.conf
MQTT_IPCSYS_DB=0
[root@GK]#
```

darkxst, 2022-09-28:
That all looks correct for the new firmware, it uses T_RecordFile. t_alarm_log is not used on new firmware at all.

Try again this command
```
[root@GK]# . /mnt/mmc/sonoff-hack/script/env.sh 
[root@GK]# mqtt-sonoff
```
You should see logging showing anytime a motion event is detected. mqtt is running that exact same sqlite command that showed your events above.

josecarre, 2022-09-29:
Hi, i did that but maybe after so much tests something else was stuck:
```
[root@GK]# killall mqtt-sonoff
[root@GK]#  . /mnt/mmc/sonoff-hack/script/env.sh
[root@GK]# mqtt-sonoff
Starting mqtt_sonoff v0.1.0
Trying to connect... Client sonoff-cam-garage sending CONNECT
Client sonoff-cam-garage received CONNACK (0)
Client sonoff-cam-garage sending PUBLISH (d0, q1, r1, m1, 'sonoffcamgarage/status', ... (6 bytes))
.
connected!
Client sonoff-cam-garage received PUBACK (Mid: 1)
^C
[root@GK]#
```
And it got stuck there, no more triggering as in previous attempts of same test...

**Finally I've rebooted the cam, and it seems to be working now...** Maybe she was not feeling fine with so much troubleshoot 😆 

I'll make some tests to check that everything goes fine, but it seems that yes and your built works with the very-last sonoff base software.

darkxst, 2022-09-29:
Was it stuck completely? even without motion events you should see keep alive messages every couple of minutes.
```
Client sonoff-cam-aa3e sending PINGREQ
Client sonoff-cam-aa3e received PINGRESP
Client sonoff-cam-aa3e sending PINGREQ
Client sonoff-cam-aa3e received PINGRESP
```

darkxst, 2022-10-04:
@josecarre it should be merged soon enough, now that I have added the setting to the webui.

josecarre, 2022-10-04:
Sounds great, thanks both 😍

roleoroleo, 2022-10-05:
This fix will be included in the next release.