roleoroleo/sonoff-hack #111: new dependency on cJSON?

darkxst, 2022-10-06:
Hi @roleoroleo, 
  I have been thinking about adding support for HA MQTT auto discovery. pulling in[ cJSON](https://github.com/DaveGamble/cJSON) would make that much easier. Are you ok with adding an extra lib into the mix. As a plus this is also used by newer versions of Mosquitto to allow the client apps mosquitto_sub/mosquitto_pub to handle JSON messages also. 
  
WIP branch that builds only the cJSON module so far is here https://github.com/darkxst/sonoff-hack/tree/cJSON. Lib seems to be pretty lightweight.

roleoroleo, 2022-10-07:
No problems.

Did you take a look at this piece of code?
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/src/static/static/yi-hack/script/mqtt_advertise

darkxst, 2022-10-12:
> Did you take a look at this piece of code? https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/src/static/static/yi-hack/script/mqtt_advertise

yeh I have seen that, I dont plan on adding any telemetry data and the sonoff cams don't have many options in comparision. Will just be motion, motion_image and maybe later a couple of controls to adjust detection settings.

roleoroleo, 2022-10-14:
Ok, no problem adding a new library.
Because the code is on the sd card and there are no problem with the size of the hack.

darkxst, 2022-10-16:
ok, I just sent through a pull request for the HA discovery stuff.

closed 2022-10-25 by darkxst