roleoroleo/sonoff-hack #116: cron in GK-200MP2-B with 0.1.3 not working as expected

josecarre, 2022-11-01:
Hi,

My camera:
[image]

Sometimes gets stuck and not work anymore... like after 3 or 4 days working... is stops showing the GUI, or video...etc, so I have to reboot it
Till i have time to troubleshoot or wait if in next FW is fixed... my workaround is to reboot from time to time... because when is stuck only way is reset unplugging and plugging from AC.

My cron:
[image]

And seems to get recorded:
```
[root@GK]# crontab -l
0 8 0 0 0 /sbin/reboot
* * * * * /mnt/mmc/sonoff-hack/script/thumb.sh cron
0 * * * * sleep 20; /mnt/mmc/sonoff-hack/script/clean_records.sh 25
```

But... it doesnt reboot everyday at 8am... anything Im missing?

Thanks!

BTW, the sonoff-hack I have is the fork from @darkxst mentioned here: https://github.com/roleoroleo/sonoff-hack/issues/108

josecarre, 2022-11-01:
Can be removed, found that my cron was not right :D

closed 2022-11-01 by josecarre

darkxst, 2022-11-01:
@josecarre I was gonna say your cron was set to fire once per month, but see you worked this out!
Please try test the latest snapshot from master here
https://github.com/roleoroleo/sonoff-hack/actions/runs/3370860439

If you are using motion detection, it has a fix to reduce the load caused by the thumbnailer.  That may resolve your hangs.

It also has the previous MQTT fixes and now has MQTT auto-discovery in Home Assistant along with some other fixes that have been added since the snapshot you are using.

josecarre, 2022-11-02:
Thanks! sure, i'll test the new release, any procedure to avoid smashing config? it would be nice to have on the UI an upgrade button to do it from local with file

btw, regarding the cron... I think is still not right on my side, I configured it through UI... result:
```
[root@GK]# crontab -l
0 8 * * * /sbin/reboot
* * * * * /mnt/mmc/sonoff-hack/script/thumb.sh cron
0 * * * * sleep 20; /mnt/mmc/sonoff-hack/script/clean_records.sh 25
```
why is not being rebooted everyday at 8am?

maybe i shouldnt have deleted the initial msg sorry, cannot get it back, but even now is not working and i thought it should be right

josecarre, 2022-11-02:
Updated... and some hours later the load is much better than usual, thanks :)
[image]

darkxst, 2022-11-02:
@josecarre 

> Update: I see this now, is from today? is that one including what you mentioned?

Yes 0.1.4 release is from today and includes all the fixes I mentioned.

> btw, regarding the cron... I think is still not right on my side, I configured it through UI... result:

Your cron job looks correct now. With the new release you can now enable syslog through the web UI after that you should see messages from cron in ```/var/log/messages```. maybe that might provide a hint ;)

darkxst, 2022-11-02:
FYI cron fires jobs in UTC time, not your local timezone.

darkxst, 2022-11-02:
> it would be nice to have on the UI an upgrade button to do it from local with file.

That would certainly be possible, but not sure we want to make it too easy for people to upload random workflow snapshots as they wont always be in a stable state!

josecarre, 2022-11-02:
> > it would be nice to have on the UI an upgrade button to do it from local with file.
> 
> That would certainly be possible, but not sure we want to make it too easy for people to upload random workflow snapshots as they wont always be in a stable state!

You're right. Maybe on a hidden `?page=` page, but yeah I agree that maybe it doesnt add much value, and just risk xD

josecarre, 2022-11-02:
Ok, just saw this :D
[image]

darkxst, 2022-11-02:
> Cannot find the usual messages file... any other place where logs are written in?

I missed something when rebasing patches, logging not working just yet. But you can copy the boot.sh from below PR and it will work
https://github.com/roleoroleo/sonoff-hack/pull/118

> Maybe on a hidden `?page=` page,

Its not that hard to update via ssh though, this is how I upgrade usually.
- Extract the tarball to `/mnt/mmc/.fw_upgrade`
- sync a copy of `/mnt/mmc/sonoff-hack/etc` (exlcuding defaults.tar.tz) into that tree
- Reboot and the final phase of the online firmware upgrade will finish it off

josecarre, 2022-11-03:
It seems that cron is now working fine :), thanks

Regarding ssh patching thanks for the procedure, that will be enoiught for me when needed, no more life risking in the ladder 😆

roleoroleo, 2022-11-03:
> > it would be nice to have on the UI an upgrade button to do it from local with file.
> 
> That would certainly be possible, but not sure we want to make it too easy for people to upload random workflow snapshots as they wont always be in a stable state!

If you check the source, there is a trick...
https://github.com/roleoroleo/sonoff-hack/blob/master/src/www/httpd/cgi-bin/fw_upgrade.sh